synman/Octoprint-Bettergrblsupport #2: Z height

Ofoten3dogcnc, 2019-05-10:
Hi, maybe Im missing something, but cant find anywhere to adjust the Z height? Only the X and Y axis :-(

thothloki, 2019-05-24:
Did you go into the plugin settings and check the box to enable Z?
there is a checkbox with "Show Z Axis on Grbl Control tab"

synman, 2019-05-25:
Let me track down a screenshot...

synman, 2019-05-25:
I'm guessing what you really want is Z axis jogging control...

[image]

Ofoten3dogcnc, 2019-05-25:
Hi :-) 
Yes I did, and it shows the Z height, butthere is no buttons to control the Z height, it just shows the current hight :-(
/Jonna
    På lørdag

synman, 2019-05-25:
I'll add some z buttons in a future release

Ofoten3dogcnc, 2019-05-25:
Ok, so there is no way to control the Z axis in this plugin now? How can I then set it to "home"? Is there any other way to do it? :-)
/Jonna

synman, 2019-05-25:
You can just plug in a z travel command in the terminal window as a work-around.  Home button currently just goes to x and y 0.  I'll add Z to it as it is no biggy.

Actual homing (exercising your actual end stops if you have them -- the targeting button) is machine specific.  It should home based on your Grbl settings.

Ofoten3dogcnc, 2019-07-17:
Hi again :-) Any progres on adding the Z height buttons? Thats the only thing missing, in an AWSOME plugin :D

synman, 2019-11-05:
not yet but I'm (slowly) gearing up for diving back in.   My machine has been collecting dust for a couple months... with it getting cold out I'm sure I'll have some cycles.

synman, 2019-11-06:
As I don't have a CNC machine / variable Z axis, I want to make sure I've got the homing portion of this correct.

If I set Z0 as part of a home move, will it drive the Z axis into the surface?   I need to understand a little better how Z relates to homing.   

I assume Z0 is calibrated / set based on the spindle being immediately above the surface to cut.

I may delay taking action on the homing aspects of this issue, but this shouldn't prevent me from getting Z travel buttons added for V1.7.0.

Ofoten3dogcnc, 2019-11-07:
Hi again :-)
Dont think there is a need to home the Z axis, at least what I feel Im missing is a way to control the axis up and down if that is possible to make? When the axis is placed where i want it, I then can home in the Gcode :-)
/Jonna

synman, 2019-11-07:
I may have this wrapped up tonight (just Z jogging).   As soon as I do, I will process a pull request for v1.7.0 and release everything tagged with the 1.7.0 milestone.

Ofoten3dogcnc, 2019-11-07:
Thanks, that would be realy awsome :D
/Jonna

synman, 2019-11-08:
How's this look?

[image]

Ofoten3dogcnc, 2019-11-08:
Perfect :-)

synman, 2019-11-08:
ok, one last question.... what GRBL command do you use today for Z jogging?   There are probably 5 different ways I can pull this off....  figure it best to go with the one that works for the requester :)

synman, 2019-11-08:
I'm just going to go with:

`G1 Z{distance} F4000`

I use a feed rate of 4000 for all the X and Y moves so I assume this speed will work fine for Z too

PR #13 (2019-11-08): Delivery of 1.7.0 milestone
Closes #2 
Closes #5 
Closes #7 
Closes #9 
Closes #11 
Closes #12
closed 2019-11-08 by synman