synman/Octoprint-Bettergrblsupport #3: Grbl Error 25

mompes, 2019-05-23:
I have just installed Octoprint and Better grbl support on a raspberry pi and I am getting the error 25 (Invalid gcode ID:25 Repeated g-code word found in block.). I have tried a couple of times and it doesn't happens at the same moment, but it happens very quickly, just 10-15 seconds after starting.
Do you have any advice about it?

synman, 2019-05-25:
What program are you using as your slicer?   If you enable logging within Octoprint you should be able to get a "dump" of the actual gcode being streamed.  I'm guessing either my parser is choking on an unrecognized set of commands or your engraver doesn't like it.

What engraver are you using?   2 or 3 axes?   More than happy to help troubleshoot... would be awesome to see your octoprint log with debugging enabled when it chokes.

thothloki, 2019-05-25:
I am trying to help him as well. I have octopi loaded with your plugin, i just need to test it.
I am using T2Laser as my slicer and will be using an EleksMaker Mana SE and a L7 board as my controller
2 axis A3 machine

What slicer do you use? @synman

synman, 2019-05-25:
Lightburn

thothloki, 2019-05-25:
> Lightburn

are you able to post an example gcode file? mgiht be something useful to be able to test if it is our slicer or hardware. Especially if it is a known working gcode file.

synman, 2019-05-25:
Provide your comm logs and Octoprint logs, ideally in debug or verbose format, if you want me to look into it.

Alternatively, you are more than welcome to fork the code, troubleshoot it yourself, and submit a pull request if you find something others can benefit from.

thothloki, 2019-05-25:
Will do. I am trying to help Mompes on this issue and was curious to compare the lightburn gcode to T2Laser gcode to see if there is much of a difference.
I have not been able to hook this up to one of my engravers yet. Life keeps getting in the way

thothloki, 2019-05-27:
@synman i will most likely have to fork this. It does not take .nc files (the file format that T2Laser generates) and i got an error pretty quick (22 i think). The only think I am thinking of changing is the allowed file formats and the gcode sending code.
I think you set yours up to specifically work with lightburn

thothloki, 2019-05-27:
I think I found the issue that I am having. Lightburn gcode is not optimized as T2Laser's is and you use the "G/M" codes at the beginning of each line to process your gcode. 
I was able to export that kind of gcode file, but it still screwed up the size of my engraving. it said a 175x151mm engraving was somewhere over 8 million mm each direction.
This is an awesome plugin, but I suggest you add a notice on your plugin that it is optimized for Lightburn and will require unoptimized gcode

synman, 2019-06-01:
upload your t2 gcode if you want me to take a look at it.

closed 2019-09-22 by synman