synman/Octoprint-Bettergrblsupport #4: Laser Controls Persistent

barnold96, 2019-09-10:
I've used Octoprint for several years on my 3D printer and was excited to see that someone finally created a plug-in to allow its use on a laser engraver and CNC.  I got my laser going just fine and it seems to operate as it should.
  
The issue I have arose when I started setting up an SD card for my CNC.  The laser controls stay regardless of how I try to manipulate the settings.  Under the 'Hide the Octoprint Control tab', there's a block to check, 'Add Laser Controls and State to Control tab'.  Regardless of how I check the blocks, the laser settings on the GRBL Control page are there.  For CNC operation, they need to go away or be replaced with something related to CNC operation.

Thanks!

synman, 2019-11-05:
I believe what you're seeing is something I left in from the original GRBL Plugin.   

"Add Laser Controls and State to Control tab" actually adds the following to the OOB control tab.  It is kinda hokey but I thought it was worth leaving in, at the least to pay homage to the original concept:

`self.customControlsJson = r'[{"layout": "horizontal", "children": [{"commands": ["$10=0", "G28.1", "G92 X0 Y0 Z0"], "name": "Set Origin", "confirm": null}, {"command": "M999", "name": "Reset", "confirm": null}, {"commands": ["G1 F4000 S0", "M5", "$SLP"], "name": "Sleep", "confirm": null}, {"command": "$X", "name": "Unlock", "confirm": null}, {"commands": ["$32=0", "M4 S1"], "name": "Weak Laser", "confirm": null}, {"commands": ["$32=1", "M5"], "name": "Laser Off", "confirm": null}], "name": "Laser Commands"}, {"layout": "vertical", "type": "section", "children": [{"regex": "<([^,]+)[,|][WM]Pos:([+\\-\\d.]+,[+\\-\\d.]+,[+\\-\\d.]+)", "name": "State", "default": "", "template": "State: {0} - Position: {1}", "type": "feedback"}, {"regex": "F([\\d.]+) S([\\d.]+)", "name": "GCode State", "default": "", "template": "Speed: {0}  Power: {1}", "type": "feedback"}], "name": "Realtime State"}]'`

synman, 2021-12-13:
I suspect there is more to this issue.  Reorganizing labels

synman, 2021-12-13:
I wonder if it is tied to #19

synman, 2021-12-19:
closing this given work done with 1.7.6.  I am creating a new milestone tonight (1.8.0) which will include bonified CNC support now that I have a CNC machine to play with too.

closed 2021-12-19 by synman