synman/Octoprint-Bettergrblsupport #5: Need Adjustable Weak Laser Setting

barnold96, 2019-09-19:
When clicking on 'Weak Laser' to focus sets laser power to S1.  For my 2.8w laser to come on, I need to be able to set it to S3.  I can do it manually, but can you make the 'Weak Laser' configurable.

lathemaster, 2019-09-22:
Also allow vale if 0 so you can see the frame and position the material to be engraved

synman, 2019-11-05:
When I add this, I'll simply make it configurable with a default value of 1.  If you want your weak laser setting to be 0 or 3, or 255 (DOH) you'll be able to just override it in the Plugin Settings panel.

synman, 2019-11-06:
[image]

PR #13 (2019-11-08): Delivery of 1.7.0 milestone
Closes #2 
Closes #5 
Closes #7 
Closes #9 
Closes #11 
Closes #12
closed 2019-11-08 by synman