synman/Octoprint-Bettergrblsupport #7: Buffer Overflow

thothloki, 2019-09-22:
Hey, I have this installed and am using an EleksMana board to control a laser engraver.

The issue I am having is that octopi with Better GRBL Support is feeding the gcode file way to fast and causing a buffer overflow. 
I am looking around to see if this is something I can fix myself. If so, I would fork it and make one dedicated to EleksMaker style machines. Most other softwares I use wait for a carriage return from grbl before sending another gcode line.

Any help is appreciated.

synman, 2019-09-22:
This plugin was developed using an ekeksmaker laser engraver as it's reference device.

Have you changed any of its defaults?

What software are you using to generate your gcode?  We may find the answer there.

thothloki, 2019-09-22:
I have not changed any defaults yet. 
I am using T2Laser for gcode generation. I think I found what might be an issue that I am going to test later today. T2 slims down the gcode to be as efficient as possible as shown below. I can can uncheck this method to add all the G01 commands back in
`G21
G90
F500
M05
G00X0Y0F500
G92X101.5Y77.5
G90
M03 S0
G01 X99 Y155 M03 S0
X99.2 M03 S255
X100.4 Y155 M03 S255
X100.6 M03 S0
X100.4 Y154.8 M03 S0
X100.2 M03 S255
X99 Y154.8 M03 S255
X98.8 M03 S0
X99 Y154.6 M03 S0
X99.2 M03 S255
X100.4 Y154.6 M03 S255
X100.6 M03 S0
X100.4 Y154.4 M03 S0
X100.2 M03 S255`

synman, 2019-09-22:
Thank you for posting the "optimized" T2 code.  Let me know how it goes with that option unselected.  I've been waiting to see what the T2 generated gcode looks like (see the invalid ticket opened previously).

thothloki, 2019-09-22:
I dealt with this issue with the GRBL support plugin years ago and gave up in frustration. That is when i found out about the optimized code.
When I posted this question, i was looking at the code and it came to me after I had posted, that the optimized code was an issue. 
What programs have you tested with? Lightburn? EleksCAM? BCL?

synman, 2019-09-22:
Lightburn is all I use.  I'm more than happy to add additional gcode style support (where feasible) but need sample gcode to feed my own machine to figure it out.

thothloki, 2019-09-22:
here is a sample of T2Laser gcode

[snoopy3.zip](https://github.com/synman/Octoprint-Bettergrblsupport/files/3639954/snoopy3.zip)

synman, 2019-09-22:
Thank you!  Guess I'm ready to hop back in and clear the backlog.

thothloki, 2019-09-22:
Ok. So i got machine code out of T2. When I send it, here is the error 
[image]
here is the gcode
[snoops3.zip](https://github.com/synman/Octoprint-Bettergrblsupport/files/3640026/snoops3.zip)

synman, 2019-09-22:
I'll send you back Snoopy if you want to share the source (assuming it is not copyrighted).

synman, 2019-09-22:
Is snoops3 without Optimization enabled?

thothloki, 2019-09-22:
[image]

it is just a line are drawing of snoopy from the Peanuts
Yes, it is the one without optimization

synman, 2019-09-22:
Very cool.  Thanks for all the details.  This is sufficient for me to engage.  Going to classify this as a bug once I'm at my PC.

thothloki, 2019-09-22:
Thank you. Please let me know. I am excited to get my 2 laser engravers and my little homemade cnc mill using this

thothloki, 2019-09-25:
Please let me know if there is any way I can help

thothloki, 2019-09-25:
> Lightburn is all I use. I'm more than happy to add additional gcode style support (where feasible) but need sample gcode to feed my own machine to figure it out.

so I tried using a lightburn gcode file and am getting the same error

thothloki, 2019-10-05:
Ok. I figured out what was causing the overflow in the gcode. I can manually edit them (eventually building a script to auto edit them for me) to make them work. I have another problem that i will open a new issue for.

synman, 2019-11-06:
Sorry for the delay... here's snoopy gcode generated from light burn

[snoopy-LB.gcode.zip](https://github.com/synman/Octoprint-Bettergrblsupport/files/3812512/snoopy-LB.gcode.zip)

FWIW, I'm working on getting the optimized T2 Laser gcode to work.

synman, 2021-12-30:
reopening this as it has remanifested in other post processors and is not limited to just the X coordinate plane.

PR #13 (2019-11-08): Delivery of 1.7.0 milestone
Closes #2 
Closes #5 
Closes #7 
Closes #9 
Closes #11 
Closes #12
commit aabe667f (2021-12-30): - fixes #7 (again) 
- added operator widget to jog pael
- check limits before jogging
- hide progressbar if full screen
closed 2021-12-31 by synman