synman/Octoprint-Bettergrblsupport #8: Lines/Vibration with Laser Plugin

barnold96, 2019-09-27:
I've used Octoprint on my 3D printer for a couple of years and was interested to see a laser plugin for it. I added the Laser plugin to Octoprint on my laser engraver and like what I see - with an exception. I was seeing artifacts in a photo I was test-printing, then saw more definite lines when printing a gradient image. I went through steps of eliminating different parts of the communication path to determine the item causing the issue.

My system consists of an RPi3 feeding an Arduino/BOB unit to drive the steppers and a laser driver module to operate the laser itself. I bypassed the RPi to start with and got smooth prints. Then, I set up two SD cards - one with Octoprint Laser and one with bCNC to further drill into the cause. It turns out that the print discrepancy only occurs when using Octoprint Laser to feed the gcode to the system. Attached is a photo showing the result using Octoprint Laser on the left and bCNC on the right.

Suggestions?

[image]

barnold96, 2019-10-23:
Is anyone looking at this issue?

synman, 2019-11-05:
I just looked, right at this moment :)

I'm guessing this is lag or something similar.   Did the octoprint based session take longer to run?  I'm guessing it did.

barnold96, 2019-11-06:
I didn't record the specific time for each run.  There was very little difference, but the Octoprint took a bit longer.  When running the gcode via Octoprint, there was a slight vibration noticeable when touching the gantry.

synman, 2019-11-08:
stupid auto-association ... the commit above has nothing to do with this.

synman, 2021-12-13:
closing this one due the sheer oddity of it.   Please touch this issue if you are experiencing similar behavior.  Otherwise, I'm going to dismiss it as a pure oddity

closed 2021-12-13 by synman