synman/Octoprint-Bettergrblsupport #9: Sample Using plugin with Inkscape NEJE extension

lathemaster, 2019-09-28:
Nothing critical - wish my Python skill were not so rusty
I created a Gcode file using Inkscape with an extension that generates a .nc file
I changed the extension to .gcode, uploaded to Octoprint and it did print correctly.
The "issue" is with the display - the values are orders of magnitude off actual
I suspect the plug-in is expecting data in a different format.
Attached is a screen shot and the gcode file I used

FYI The laser is a NEJE Master 3500 version 5 firmware which I believe is GRBL 0.8c
Laser is 3500 mw

[neje-3500-Gothic Test-Ver2_0003.nc.txt](https://github.com/synman/Octoprint-Bettergrblsupport/files/3665947/neje-3500-Gothic.Test-Ver2_0003.nc.txt)
[image]

synman, 2019-11-05:
I think this is positioning mode related.   I'll take up the charge and thank you for the sample gcode to play with.

synman, 2019-11-05:
I see G90 at the top of your GCODE.   This tells me you are using absolute positioning.  Head over to the "Reported Sizes" issue as there is discussion about this happening there.

synman, 2019-11-06:
something else that is causing your calculations to be all wrong is the use of arcs.

PR #13 (2019-11-08): Delivery of 1.7.0 milestone
Closes #2 
Closes #5 
Closes #7 
Closes #9 
Closes #11 
Closes #12
closed 2019-11-08 by synman