synman/Octoprint-Bettergrblsupport #12: Support for Feedrate with Decimals

AlligatorFate, 2019-10-29:
I think there's a bug on line 454 of __init__.py in better grbl support for octopi. 

The observed behavior in the interface was a report of error 33 from grbl for a G02 arc command.  That error means the target destination point was not on the described arc.  The gcode was valid and the target was within the expected margin of error for grbl.  However, the start point was not the expected start point because the prior line of G01 gcode was skipped. 

The prior line of gcode was skipped as invalid because the feedrate was a float rather than an interger.  Here's an example: 

G0 X17.192318 Y22.904479 Z-0.500000 
G1 X17.834666 Y21.827774 Z-0.500000 F360.00
G3 X16.489131 Y22.201292 Z-0.500000 I-2.069250 J-4.843830 F360.00

The first G1 line causes an error like this in the octoprint log file: 

2019-10-28 22:33:37,078 - octoprint.util.comm - ERROR - Error while processing hook bettergrblsupport for phase sending and command G1 X17.8347 Y21.8278 Z-0.5000 F360.00:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3220, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bettergrblsupport/__init__.py", line 454, in hook_gcode_sending
    grblSpeed = int(match.groups(1)[0])
ValueError: invalid literal for int() with base 10: '360.00'

so the G1 line is never sent to grbl.  That means the start point remains at X17.1923, Y22.9045 rather than moving to X17.8345, Y21.8278.  And the point is not on the arc for the G3. 

Should we not change the feedrate interpretation to match the speed interpretation since the feedrate in gcode can legitimately be a float: 

So use this: 

                grblSpeed = int(float(match.groups(1)[0]))

instead of this: 

                grblSpeed = int(match.groups(1)[0])

so that we match something like this, which seems to work: 

                grblPowerLevel = int(float(match.groups(1)[0]))

 Thanks.  I haven't tried the change yet, but I intend to soon and I'll update the post with my result.

AlligatorFate, 2019-10-30:
I can confirm that the change above corrected the problem.  I did find another spot in the same file with the same problem (take a look at lines 588 and 589, too).

I haven't checked it in or anything.  Let me know if it's better to try and make a change in github for you. 

I've still got an issue with comments in the gcode causing comms problems, but if I remove the comments manually, it works.  I don't know whether that's Bettergrblsupport.  Doesn't seem like it would be. 

Thanks for creating this plugin!

synman, 2019-11-05:
ok, I this issue answers my question to you in the reported sizes issue.

Stupid python.   I agree, this is a good (solid) fix.

synman, 2019-11-05:
regarding comments, we probably just drop them in hook_gcode_sending()

go with something like this:

```
        # suppress comments
        if cmd.upper().lstrip().startswith(';'):
            self._logger.debug('Ignoring comment')
            return (None, )
```

We may need to get more sophisticated as GRBL also support inline comments.  See here for an example (taken from Universal GCode Sender):  https://github.com/winder/Universal-G-Code-Sender/blob/master/test_files/comments.gcode

Some fancy regex should cover the inline ; and () use cases.

AlligatorFate, 2019-11-06:
Yeah, I saw that about inline comments.  I selfishly checked in my change, though, because it was always blowing up in a confusing way and inline comments seemed less likely.  But a better way would be better.. :) 
Interestingly, it wasn't the ";" comments that caused a problem.  It was only the parenthetical comments.  I don't understand why.  I think the problem is that the parenthetical comment does/doesn't generate an "ok" and octoprint does/doesn't expect the "ok".  So the acks get out of sync with the lines being sent.  That's what it looked like to me without certain proof of it.  I suppressed them and moved on. 
(Unfortunately, FreeCAD generates those parenthetical comments for operation changes, and I can't see a way to configure it not to.)

synman, 2019-11-06:
quick question as I integrate this into the plugin...

```
(first comment)
G21 (inline comment 2)
```
you are merely filtering out the first example, not the second, correct?   I was thinking about squashing any lines that begin with ( or ; as my first step.

AlligatorFate, 2019-11-06:
Yeah, that’s right.  The semicolon comments were already filtered out.  I just added the same logic to filter out lines that start with “(“ like your first example.  The second line would not be filtered by my addition.  
My code addition looks just like what you’re suggesting: 

       # suppress parenthesis comments 
        if cmd.startswith('('):
            self._logger.debug('Ignoring %s', cmd)
            return (None,)

PR #13 (2019-11-08): Delivery of 1.7.0 milestone
Closes #2 
Closes #5 
Closes #7 
Closes #9 
Closes #11 
Closes #12
closed 2019-11-08 by synman