synman/Octoprint-Bettergrblsupport #18: Can't revert back to how my octoprint worked before installing this plug

Thisismydigitalself, 2019-11-28:
**Installed to check this plug. now after me uninstalling this plug my octoprint is acting weird and i don't have my gcode visulizer anymore. my optoprint plugin folder is now grayed-out as well.
I need my octoprint the same as it was before installing this plug.**

synman, 2019-11-28:
Check the readme:

https://github.com/synman/Octoprint-Bettergrblsupport/blob/master/README.md#configuration

Thisismydigitalself, 2019-11-28:
my mistake for installing this plug without reading through and backup my config.yaml
what a mess.

Never again

synman, 2019-11-28:
Ask Gina to provide plugins an uninstall sink and I'll do the cleanup for you.

closed 2019-11-28 by synman

Jason55904, 2022-03-08:
Is there any documentation on cleanup after uninstalling? I did read the documentation at the above link but didn't see see any warnings about changes that wouldn't be reverted after uninstall or see a warning to backup my configurations before trying to use this plugin. I do like your plugin but need to be able to use my printer as both a printer and a laser.

synman, 2022-03-08:
What version of Octoprint are you running?  It sounds like the plugin uninstall hooks are not firing on your instance.

Jason55904, 2022-03-08:
Thank you very much for your reply. I'm currently on version 1.7.3.

synman, 2022-03-08:
What version of Octoprint are you running?  It sounds like the plugin uninstall hooks are not firing on your instance.

Weird.  Octopi .18?

Jason55904, 2022-03-08:
Correct 0.18.

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/98.0.4758.102 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: True
connectivity.enabled: True
connectivity.online: True
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: True
env.hardware.cores: 4
env.hardware.freq: 1500.0
env.hardware.ram: 3959984128
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: True
octoprint.safe_mode: False
octoprint.version: 1.7.3
systeminfo.generator: zipapi