synman/Octoprint-Bettergrblsupport #19: Laser Mode Enabled after every Draw Frame action

AlligatorFate, 2019-12-18:
It appears to me that laser mode ($32=1) is enabled whenever I draw the frame by pressing the "Draw Frame" button.  Draw Frame functionality is useful to visualize the boundaries, but I'm using CNC, not Laser.  I never want laser mode enabled. 

The problem with laser mode while using CNC is that the "spindle" won't turn on until the cutting move (G1 as opposed to G0) is acted on.  But a real spindle needs time to spin up and can't cut immediately.  So, this causes problems.  If I set $32=0 in grbl settings, I'd like it to stay that way. 

It looks to me like drawing a frame is trying to make sure laser mode is off and then turning it back on when it's done.  However, I believe it should only do this if laser mode was on to begin with.  Otherwise, it'll turn laser mode on when it was initially off, which can be very hard to troubleshoot if you don't know where it's coming from.  Who would expect that "draw frame" would turn laser mode on when it was off before? 

(And thanks again for your work, synman.)

synman, 2019-12-19:
Remember, I only have a laser :)

Will take a look at this and see about cleaning up.

Shell

AlligatorFate, 2019-12-19:
I think my tone was wrong... sorry.  I meant to be persuasive, not critical.  It's all about continuous improvement, right?

synman, 2021-12-14:
I'm working on factoring out system ($) command interactions as much as possible right now, focusing initially on removal of all $32 commands within the plugin (except for when you want to change them via Grbl Settings).

synman, 2021-12-16:
Tying this together with #35 and #4 

This is turning into an extensive, but very necessary, exercise. I've brought a lot of stuff into scope here including laser detection based on $32 value, usage of Grbl 1.1 jogging for framing operations, and backwards compatibility for older devices.

I'd rather just drop support for pre 1.1 firmwares but recognize there are probably a bunch of people stuck on .9.

I'm also revisiting all embedded gcode and refactoring many of the commands used by the plugin today for various operations.

Lastly, I've finally, just moments ago, pulled trigger on a budget CNC machine. I went with this bad boy: https://www.foxalien.com/collections/cnc-router/products/cnc-router-machine-masuter-with-300w-spindle-bundle-kit

So yeah, anticipate some major improvements over the next month or so as I adapt the plugin to cover all my own CNC needs.

Shell

synman, 2021-12-19:
this should be fixed.  please comment if you observe otherwise

closed 2021-12-19 by synman