synman/Octoprint-Bettergrblsupport #21: Macro buttons

Gamdiiii, 2020-05-06:
hey
could you add some buttons that could be set by user to add a gcode for probing etc. 
just like the plugin: terminalcommands

conquillacarl, 2021-11-20:
Perhaps I can help here since I have had a poke around in the code now. I think a quick fix would be to have a couple text fields for Gcode macros and then a button next to them to inject them. It wouldn't look amazing but it might be convenient for someone's 5 or so most used macros.

Anyone interested?

coloradotrailboss, 2021-12-27:
Just wanted to comment on the plug-in.
Oddly enough I'm using for CNC and not my 3D Printer.
Last week the Z value was showing up on the Mode, State, X, Y, <Z> gone missing :-(
Also the issue is when I press the set Zero location, it sends a G28.1 command the the CNC isn't happy.
Currently the walk around is to use the Set Location under the Laser option which sends a G92 X0 Y0 Z0 command and then it all good.

Great work, awesome plug in.
I can n ow go straight from Fusion360 to octoprint and upload the gcode files for the CNC. No other software required.
The CNC is SaintSmart 3018, which now a 3040

Almost achieved the goal straight from Fusion360 to devices CNC or 3D printer, each device has dedicated raspberry pi running octoprint.

synman, 2021-12-27:
I'd like to understand better how G28.1 is interfering with your workflow.  What version of GRBL are you running?

It's about to get a whole lot better.  If you're interested, switch to the Release Candidate channel and you can check out what is coming with v2.0.0.   These are currently not real RC's.... I should be classifying these as dev builds, but would appreciate any feedback.

coloradotrailboss, 2021-12-27:
Hi Shell,

I just upgraded to the release candidate and updated my comments.
It's perfect!! Ran a quick test as I know what I looking for.

I'll test it with the printer again as well, it was not playing nice with octoprint dashboard for some odd reason.

Evan

coloradotrailboss, 2021-12-27:
I would happy to test for you...

If it helps, here's a default workflow for CNC.
You have to first 'Home' the machine before doing anything else.
Then move to controls in the center like you have them, being able to move X,Y on angle would be great as that is how we center the piece when working from the middle. Set origin we need for the craving.
Next, we'll run the z-probe for the offset or the old paper trick to set Z.
Being able to X,Y,Z all at once or one at a time is important. mostly because we'll do a tool change and need to reset Z, but not the X,Y.

Maybe scope some ideas from the gSende: gSender 1.0 is now here. Simple and powerful control of GRBL CNCs. | Sienci Labs<https://sienci.com/2021/11/02/gsender-1-0-is-now-here-simple-and-powerful-control-of-grbl-cncs/>
I was running this before switching over to octoprint. Didn't plan to use octoprint for the CNC, however, your plug-in made me switch. LOL.

I know have stuff in Fusion360, with two setups one for Mill one for Additive. Just cause...
Tomorrow we'll fire up the laser and see how that goes.

Have a great evening, nice work.

Chat soon,
Evan
***@***.******@***.***>

coloradotrailboss, 2021-12-27:
One other thing for the pile..
Being able to adjust feeds and speeds is important.. lost a few bits and generated tons of smoke too ☹

coloradotrailboss, 2021-12-27:
Here's the link to Spaghetti Detective
AnyCubic Chiron: https://app.thespaghettidetective.com/printers/share_token/3619437fe711331b8529b819804c802c8eb6/
SaintSmart 3018/3040: https://app.thespaghettidetective.com/printers/share_token/43885fd9a3cb504e261b7b0da55264a70515/

synman, 2021-12-27:
I have the basics workflow wise down having been driving a laser for the past year or two.   Carving has really taken it to the next level.  I'm so glad I bought myself a CNC machine for xmas.   Most of my time away from work this past week has been spent building / tweaking the machine, setting up a new PI / octopi / etc for it, etc, and of course major work happening with bgs.

You have given me a ton of valuable information here.  I'll break it out into a backlog tomorrow.  I already have some ideas for diagonal moves and individual axis homing UI wise.

I'm using VCarve for slicing.  Have a job running now carving a 3d model.   Most of my testing happens on the laser and I try to get at least one job in on the CNC before releasing a new candidate.

coloradotrailboss, 2021-12-27:
Looks like the last message was too large to send through

Maybe you could send me a different email address 😉
Not sure which time zone you are in, I’m in Mountain.
So the old guy has to go have a nap now.

If you wanted to chat on Teams (aka the old skype) I would happy to chat about how I can help and give back a little to the community.
It’s only natural that you start with one 3D printer or CNC then end up with the other one or other because they are just plain fun to play with.

Cheers,
Evan

coloradotrailboss, 2021-12-27:
Not sure the 4521.82 is possible on my little machine, a great upgrade tho.

***@***.***

synman, 2021-12-27:
you can contact me directly here:  shell at shellware.com

synman, 2021-12-31:
closing this as I believe the requested functionality has been delivered, just not via macros.

closed 2021-12-31 by synman

coloradotrailboss, 2021-12-31:
Dude, the stuff you put together over the last couple of days is pretty much everything we need on the CNC side.
Totally usable, just an instruction page as some of the features are not intuitive enough some, like me LoL.

Just repositioned the printer, so I'll pull the new code and give it a test run.

Closed #21<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fsynman%2FOctoprint-Bettergrblsupport%2Fissues%2F21&data=04%7C01%7C%7Cd27538fa90274301b90d08d9cc9c9b2e%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637765795254509017%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=zvscFMrOLPRwe7CTuwIFDZU6khOmRRpJ%2BOmosBPa2Cc%3D&reserved=0>.

-

synman, 2021-12-31:
very valid callout.... i'll create a documenation issue.... will likely build out a wiki.

here's everything that went in to v2.0.0:

https://github.com/synman/Octoprint-Bettergrblsupport/releases/tag/2.0.0

coloradotrailboss, 2021-12-31:
FYI: changed the name and model on default CNC and things worked fine.

Just a ton of work, felt bad bugging you.

How many users do you have using it?

https://github.com/synman/Octoprint-Bettergrblsupport/releases/tag/2.0.0<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fsynman%2FOctoprint-Bettergrblsupport%2Freleases%2Ftag%2F2.0.0&data=04%7C01%7C%7Cf93c5560a72e4273107608d9cc9e2ab1%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637765801958244891%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=BFNn7K%2Fk1kD3cqnjzZWBWwVxsRmyj3JxY0wbZJ15cWc%3D&reserved=0>

-

synman, 2021-12-31:
According to the plugin page:

> Active instances the past month
> Installed on at least 225 instances
> 
> New installs the past week
> At least 26 new installations

coloradotrailboss, 2021-12-31:
Once the word gets out that you support CNC machines I'm sure business will pick up.

For my configuration use the following:
Started with SaintSmart 3018 CNC, which is now a 3040 CNC with a laser.
A new addition this year aka Christmas an Anycubic Chiron

For software I have used Easel, then Fusion360, Craveco most of these focused on the CNC side of things.
With the addition of Chiron I wanted a software package that would do CAD/CAM which brought me back to Fusion360
With Fusion, I can do my design work for the 3D Printer, CNC, and Laser and output the gcode.

Using Octoprint and Better GBRL plug-in I can now drive both the CNC and the 3D Printer.

Thanks, Shell for all the efforts over the last couple of days to get things dialed in.
Happy to help out in the future, just let me know what you need.

ColoradoTrailBoss

synman, 2022-01-01:
I use Fusion360 and tinkercad quite a bit for modeling.  I use OpenSCAD for some stuff too.

I recently dropped S3D (after years of using it)  for SuperSlicer for slicing stls on the printer.  

Everything for my laser gets sliced with Lightburn.

I didn't bother testing anything for CNC.  I went straight to VCarve Desktop.  I haven't bothered trying other CNC slicers, yet.

I used my first 1/4" bit tonight... a 3mm ball nose for a rough 3d cut :)

coloradotrailboss, 2022-01-01:
You need an end mill for a rough cut, at with ball nose it will look good when it's done 😉

Learned a long time ago the closer you get to a single platform (app & o/s) the better off you are.

As much as SGI machines were a pain, they were always solid, everything was designed to match.
when I started doing that with Windows, guess what no more blue screens.

Fusion does all three, CNC, Laser and 3D Printing, so only need one app, one subscription to get the benefits of the add-on features.
OctoPrint and BG to drive the machines, I have a Raspberry Pi for each machine.
For the printer, I load Octoprint o/s, so it's clean.
The Pi for CNC does double duty, OctoPrint is installed in Portainer sitting on top of Docker.
I use Docker to run both Octoprint and Home Assistant for home automation. It’s also the machine that gets rebooted most often.
Currently its running Bullseye as the o/s

Crazy as it sounds I was doing all the reconfiguration while we were testing, what could possibly go wrong LOL

What a great way to end the year, Printer is up and running, CNC is off and craving thanks to BG

synman, 2022-01-01:
I didn't like how much my various end mill bits were leaving left over for the final cut so I modeled it with a 3mm ball and bingo.   detail cut is with a 1mm ball.

[image]

coloradotrailboss, 2022-01-01:
Ah, that would be the two-pass theory we use.
Roughing pass, as deep and as fast as you can go, sometimes slower is better you want to remove as material as possible big step over as well. leave about .02 mm stock for the detailed pass.
Then detailed pass, usually a ball nose or other detail bit should only need a single pass, very tight step over to reduce any sanding needs. It should pick up from the last toolpath as the starting point knowing it only has about .02 mm to remove.

Hence why you are getting two files from me.
Normally I use three.

  1.  Facing 22mm flat bit or I use a surfacing routine – this to flat boards both sides 😉
  2.  Roughing Pass normally 3.175 end mill, .5 step down, 40 % step over, don’t care about ugly, just grip and rip the stock out.
  3.  Parallel Pass  X (maybe two Y) currently using r0.025 bullnose pretty much the point of pencil only sharper 😉

[10a0412cee973265b8ea702ccdc920f5524c94c74b9423cfdf949516f2f727f8]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F1299716%2F147845818-2291674a-8e07-41ba-9580-585f37341583.jpg&data=04%7C01%7C%7C5bff645509cd4a6b460808d9ccf6b73d%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637766182269917414%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=Gmc7h0YQK0LvrBPDO5ZPoPjCHpneSguh%2FAcpiiUxVgM%3D&reserved=0>

synman, 2022-01-01:
I think you are confused.  that's exactly what I did.  I used the bigger 3mm ball for my rough pass and the 1mm for the final / detail one.

in VCarve the end mill bits that I have did not dig all that deep into material leaving a ton of work for the detail run.  So I said screw it and used a 3mm ball for the rough and yeah, the 1mm ball made one pass and it is now smooth as silk... not that it already wasn't pretty damn smooth with 3mm ball

coloradotrailboss, 2022-01-01:
LOL…

You could have just left the end mill and pretended it was 3mm ball nose.. the machine doesn’t know what bit is in there 😉

Check the step-down and the multi-pass is turned on for the end mill
I’m going through again as I convert my stuff from Carvco back to Fusion, plus I have to put in the new bits I have…
And re-test all the feeds and speeds to make sure they operate as expected.

The good news is the BG gCode processor is working.
So now I only have to deal with Fusion set up.