synman/Octoprint-Bettergrblsupport #23: Expected Command Letter Failure

ShadowCory, 2020-05-26:
Hi, I am trying to run a pen plotter with a Ramps 1.3 board, GRBL for Ramps, and using Inkscape and Gcodetools to generate the g code. I can control the plotter with the Universal Gcode Sender, but I can't seem to be able to send a file over and plot it with this Octoprint plug in. I can jog all three axis as much as I like, but sending the gcode file gives me the following:
"
Recv: ok
Send: ?$G
Recv: <Idle,MPos:0.000,0.000,0.000,WPos:10.000,30.000,60.898>
Recv: [G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0.]
Recv: ok
Changing monitoring state from "Operational" to "Starting"
Changing monitoring state from "Starting" to "Printing"
Send: %
Recv: error: Expected command letter
Send: M3
Changing monitoring state from "Printing" to "Error: error: Expected command letter"
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Changing monitoring state from "Error: error: Expected command letter" to "Offline (Error: error: Expected command letter)"
Connection closed, closing down monitor
"
Here is the gcode I am trying to print:
%
(Header)
(Generated by gcodetools from Inkscape.)
(Using default header. To add your own header create file "header" in the output dir.)
M3
(Header end.)
G21 (All units in mm)

(Start cutting path id: path595)
(Change tool to Default tool)

G00 Z35.000000
G00 X-491.779890 Y-122.049310

G01 Z-1.000000 F800.0(Penetrate)
G02 X-490.116910 Y-125.322640 Z-1.000000 I-0.343640 J-2.233677 F800.000000
G02 X-492.648940 Y-127.698140 Z-1.000000 I-4.631748 J2.399763
G02 X-499.796749 Y-126.487713 Z-1.000000 I-2.654200 J6.036252
G02 X-503.077560 Y-120.311210 Z-1.000000 I7.295371 J7.834722
G02 X-499.430246 Y-109.549923 Z-1.000000 I12.101184 J1.897290
G02 X-489.172740 Y-105.102810 Z-1.000000 I11.170787 J-11.712721
G02 X-474.733751 Y-111.086223 Z-1.000000 I1.015228 J-17.963653
G02 X-469.184560 Y-125.525510 Z-1.000000 I-16.306012 J-14.552550
G02 X-477.472486 Y-143.668194 Z-1.000000 I-23.732349 J-0.122992
G02 X-496.125130 Y-150.293480 Z-1.000000 I-17.935954 J20.926672
G02 X-517.984062 Y-139.714657 Z-1.000000 I0.766229 J29.456126
G02 X-525.672890 Y-116.835020 Z-1.000000 I25.550664 J21.318197
G02 X-512.810249 Y-91.252855 Z-1.000000 I35.155714 J-1.651462
G02 X-485.696550 Y-82.507480 Z-1.000000 I24.699480 J-30.173337
G02 X-456.386877 Y-97.649806 Z-1.000000 I-2.533458 J-40.841038
G02 X-446.589220 Y-129.001700 Z-1.000000 I-34.794026 J-28.080205
G02 X-464.008627 Y-162.041697 Z-1.000000 I-46.517274 J3.412977
G02 X-499.601330 Y-172.888810 Z-1.000000 I-31.460554 J39.412880
G02 X-536.373594 Y-153.194088 Z-1.000000 I4.290612 J52.187397
G02 X-548.268220 Y-113.358820 Z-1.000000 I44.030192 J34.840685
G02 X-526.299424 Y-72.852883 Z-1.000000 I57.853254 J-5.166820
G02 X-482.220350 Y-59.912150 Z-1.000000 I38.220659 J-48.646209
G02 X-437.979696 Y-84.154109 Z-1.000000 I-6.041905 J-63.515983
G02 X-423.993890 Y-132.477900 Z-1.000000 I-53.261231 J-41.600566
G00 Z35.000000

(End cutting path id: path595)

(Start cutting path id: rect10)
(Change tool to Default tool)

G00 Z35.000000
G00 X-330.136360 Y-85.549170

G01 Z-1.000000 F800.0(Penetrate)
G01 X-112.873540 Y-85.549170 Z-1.000000 F800.000000
G01 X-112.873540 Y-125.525530 Z-1.000000
G01 X-330.136360 Y-125.525530 Z-1.000000
G01 X-330.136360 Y-85.549170 Z-1.000000
G00 Z35.000000

(End cutting path id: rect10)

(Footer)
M5
G00 X0.0000 Y0.0000
M2
(Using default footer. To add your own footer create file "footer" in the output dir.)
(end)
%

Thanks for any pointers you can give me.

synman, 2021-12-13:
I'm going to filter out this character (line) when it shows up at the beginning of a line.

synman, 2021-12-13:
so % is definitely part of the GRBL protocol.  It signifies beginning and ending of a program (series of gcode commands).  technically this is working as designed and the problem is on the ramps end of this.

commit 46f0e41b (2021-12-13): fixes #23
closed 2021-12-19 by synman