synman/Octoprint-Bettergrblsupport #26: Smoothieware does not support GRBL config commands

hgroover, 2020-08-23:
I am running a controller board with the Smoothieware firmware. It rejects the $$ command (and a few other GRBL commands; see http://smoothieware.org/from-grbl?s[]=grbl&s[]=configuration&s[]=commands this list). This plugin seems to offer a lot of useful functionality and the Smoothieboard is not a bad CNC controller. Unfortunately it raises an error on $$ which causes Octoprint to disconnect. Admittedly it is a somewhat knuckleheaded approach in Smoothieware to do a few things differently and not support aliases. I'm reporting this as an issue not because it's a flaw in this plugin but because I believe others may try this with Octoprint + Smoothieware.

hgroover, 2020-08-24:
I cloned develop and changed a few lines to support Smoothie and it all seems to work locally (except the config parsing, I didn't expect that to work but it doesn't raise an error). The documentation from the Smoothie project was incorrect, cat /sd/config.txt is the correct substitute for $$.

Working on adding the checkbox(es) to enable this behavior into the configuration UI and running more tests, and I can do a PR when I'm done if you're interested. This is a really great plugin and very useful! Currently I've been suffering through the Smoothie native web interface which is not always responsive and doesn't handle large file uploads.

synman, 2021-01-21:
if you share your code, either here, or via pull request I'm more than happy to add this to the master branch

commit cb22908e (2021-01-21): add (experimental) smoothieware support - closes #26
closed 2021-01-21 by synman