synman/Octoprint-Bettergrblsupport #27: Separate zero-ing X, Y, Z axes

conquillacarl, 2020-09-08:
Thanks for your awesome work! Is there a plan to have separate X, Y, Z axis zeroing? When working with really tiny tools it can help to zero the axes independently (unless I'm doing it wrong!) 

Kind thanks

synman, 2021-01-21:
I am definitely not a critic on how to manage your workflow... I'm much more inclined to say figure out what works for you and go for it.

I can look into axis specific zeroing.   One trick will be figuring out how best to integrate it into the UI.

Willmac16, 2021-02-15:
Are you talking about homing one axis at a time or something else? If you are talking about that, it needs to be enabled in the Grbl config.h file and can be done by running $HX, $HY, or $HZ, or you could change the homing routine to do one axis at a time.

synman, 2021-02-16:
I read the ask to be zeroing the relative axes individually rather than doing an actual homing operation.

conquillacarl, 2021-02-16:
Yes exactly, as in being able to zero X and Y axes separately when your Z height is not at zero. But that's actually also quite useful to know about the independent axis homing especially if it works with gantry configurations

conquillacarl, 2021-11-19:
Hey @synman I'm interested in helping out here if I can. I've just fired up this CNC after being away for a bit and this feature would really help. 

I was also hoping that there could be separate "move to home" buttons for the XY and the Z axes since you don't always want your tool plunging to the height of your stock. 

As for the UI you could just do it the same way that Roland does it. You have at the moment two button groups: XY and Z, you could have a zero button in the top right of each group and the move to home in between the jog arrows.

closed 2021-12-07 by conquillacarl