synman/Octoprint-Bettergrblsupport #31: New install error - IndexError: list index out of range

sspitz01, 2021-01-20:
I started a new install of octo print on a Raspberry 4 to run my Laser engraver. the plugin better grbl support is not working. I used  OctoPi 0.17.0 image full image and the newest 0.18.0 image.  

octoprint version Version 0.17.0, running on Raspberry Pi 4 Model B Rev 1.1

below is the error I am getting. 
2021-01-20 20:28:41,363 - octoprint.plugin - ERROR - Error while calling plugin bettergrblsupport
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 219, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bettergrblsupport/__init__.py", line 241, in on_after_startup
    self.deSerializeGrblSettings()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bettergrblsupport/__init__.py", line 282, in deSerializeGrblSettings
    self.grblSettings.update({int(set[0]): [set[1], self.grblSettingsNames.get(int(set[0]))]})
IndexError: list index out of range

synman, 2021-01-20:
From 20,000 ft I don't think this has anything to do with your fresh install....

```
File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bettergrblsupport/init.py", line 282, in deSerializeGrblSettings
self.grblSettings.update({int(set[0]): [set[1], self.grblSettingsNames.get(int(set[0]))]})
IndexError: list index out of range
```
Indicates the plugin is having a hard time iterating your GRBL firmware settings.   Has it ever worked with your engraver?

synman, 2021-01-20:
What version of GRBL is on your engraver?

Assuming your engraver is actually connected, compare your "$$" results with the list provided here:  https://github.com/synman/Octoprint-Bettergrblsupport/blob/master/octoprint_bettergrblsupport/static/txt/grbl_settings.txt

sspitz01, 2021-01-21:
GRLB of mine is 1.1 I think f or g. 

The grbl_settings.txt are not showing up in the plugins area? This is what caused me to trace down the errors in the log. My laser will not connect I keep getting a errors below. Not sure what i am doing wrong? 

terminal output from when I try to connect. 
I connect trough other software and was able to send a M5 and it worked. 

Serial port list: ['/dev/ttyUSB0']
Connecting to: /dev/ttyUSB0
Changing monitoring state from "Detecting serial port" to "Opening serial port"
Connected to: Serial<id=0xaa93cbb0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 115200
Send: M5
Recv: error:7
Changing monitoring state from "Detecting baudrate" to "Error: error:7"
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Changing monitoring state from "Error: error:7" to "Offline (Error: error:7)"
Connection closed, closing down monitor

synman, 2021-01-21:
```
Error 7
EEPROM read fail. Using defaults An EEPROM read failed. Auto-restoring affected EEPROM to default values.
```

This is coming from your laser's control board.

documented here:  https://github.com/synman/Octoprint-Bettergrblsupport/blob/master/octoprint_bettergrblsupport/static/txt/grbl_errors.txt

synman, 2021-01-21:
I found this when doing an omniscient google search:

> grbl stores its parameters (the ones you posed above) in the ATmega328P EEPROM. When retrieving theses parameters if there is an error then grbl uses the defaults that were loaded in with the firmware.
> 
> So, there is a problem reading the EEPROM on your ATmega328P. If you know how to use the Arduino IDE you can run the example program to clear the EEPROM and start over.
> 
> A reload of the firmware might take care of the problem, maybe not.

While using your "other" software, what software is that specifically?  What happens if you type "$$" in that other software's terminal window?

synman, 2021-01-21:
hmm.... while I never intended for the plugin to be used as part of initial system setup I'm going to guess, at the very least you have not saved / initialized your built-in GBRL settings ($$).   Your control board is barfing because it is trying to read them from its internal storage (EEPROM) and it is not finding them.

I can probably ignore this error on connection, or treat it as a warning, but you really do need to get your GRBL firmware settings saved on the actual EEPROM.  This usually involves some configuration on your part as a ton of settings critical to performing successful operations come from this storage area.

sspitz01, 2021-01-21:
I was using lightburn. now I see you are pull the grlb_settings from the printer.  that is why it does not show yet. 

Below is what i am getting. 

[VER:1.1f.20170801:]
[OPT:VI,15,128]
Target buffer size found
ok
$$
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=80.000
$101=80.000
$102=250.000
$110=6000.000
$111=6000.000
$112=1000.000
$120=1000.000
$121=1000.000
$122=10.000
$130=410.000
$131=400.000
$132=200.000
ok

I also noticed that when lightburn connected I see these errors
Waiting for connection...
error:7
Waiting for connection...
error:7
EEPROM read failed. Reset and restored to default values.

sspitz01, 2021-01-21:
ok so the values are not stored correctly in the controller board then. do I need to recompile the grlb with the values and re flash it? I figure the manufacture would have done this but you never know.

synman, 2021-01-21:
Like the guy I quoted above, I really have no idea.  I can only guess what root cause is based on the information you feed me.  I'm working on a couple things tonight... One is a general "ignore errors" option.   

Try this and see if the error goes away:

1.  copy everything above from $0 through $132 to your clipboard
2. paste it back into the Lightburn terminal window
3. disconnect and reconnect

I wonder if forcing a write of each setting will clear the error.

sspitz01, 2021-01-21:
ok thanks. I was also making some changes to the error checking loop to ignore the error to see what happens. 

I will try it in the morning and see if that help. 
Thanks!!!!!

synman, 2021-01-21:
you'll find I have the constructs for it already.... look for self.ignoreErrors.  I'll have a release candidate available in a bit that has it exposed as a configurable setting in the UI in an hour or so.

synman, 2021-01-21:
guess we can call this an enhancement request :)

closed 2021-01-21 by synman

sspitz01, 2021-01-21:
I installed the updates and checked the box ignore firmware errors but I still could not connect. still getting error 7.
I also flashed the controller to 1.1h from 1.1e and now it works with out checking the ignore firmware error box. 
So it was really the incorrect firmware for your plugin not your software. 

I am doing my first etch using octoprint now. 

Thanks!!!!

commit 38c584ef (2021-01-21): fixes #31