synman/Octoprint-Bettergrblsupport #35: Framing Issue

criticallimit, 2021-05-06:
Have a error trying do do Framing. Have a Neje Laser Master 2.

Send: M4 F1000 S1
Recv: ok
Send: G4 P0
Recv: error:2
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

Somebody here who knows what exactly the problem is and how to solve it?
Suspect the G4 P0 causes this Fault. But where can I change commands for Framing?

Thanx
Dirk

criticallimit, 2021-05-06:
even when I comment out all in the code, he starts framing and stops after a few mm travelling:

Send: $32=0
Send: M4 F1000 S1
Recv: ok
Send: $32=1
Send: G1 X100.000000 F2000 S1
Recv: error:2
Changing monitoring state from "Operational" to "Error"

But when I put the G1 X100.000000 F2000 S1 in the command line in the terminal, it is moving correctly

What do I have to change ?

criticallimit, 2021-05-06:
ok. here is another guess:

    def send_frame_init_gcode(self):
        # self._printer.commands("G4 P0")
        #self._printer.commands("$32=1")
        # self._printer.commands("G00 G17 G40 G21 G54")
        self._printer.commands("G90")
        # self._printer.commands("M4")
        # self._printer.commands("M4 F1000 S{}".format(self.weakLaserValue))
        # self._printer.commands("G91")
        # self._printer.commands("M8")

No matter what the command in the frame_init part. After executing, it receives a Fault

criticallimit, 2021-05-08:
Slowly I think it´s a timing Problem. No matter which command is given, it gives error2 after the first "G..." command

I´m  a totally noob in programming. Can someone tell me what to change in the code?

Recv: <Idle|MPos:50.000,50.000,0.000|FS:0,0>
Recv: [GC:G0 G54 G17 G21 G90 G94 M5 M M9 T0 F0 S0]
Recv: ok
Send: $32=0
Send: M4 F1000 S1
Recv: ok
Send: $32=1
Send: G0 X10.000000 F2000 S1
Recv: error:2
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

synman, 2021-05-14:
What version of GRBL are you running?

error:2 is a number format error

`2 | Bad number format | Missing the expected G-code word value or numeric value format is not valid.`

I'm going to guess It is barfing on the X value being passed in.

criticallimit, 2021-05-14:
It's GRBL 1.1f

I checked already the gcode. All files are tested where the same as direct connected to the computer. The gcode in your bettergrbl has more digits after the point. Can I reduce it to 2 or 3?

criticallimit, 2021-05-14:
But even if there is only a G90 is sent, there is error2

Tennissocke, 2021-05-24:
I have the same problem with the Neje Master 2s. everything works, only when I want to drive off the frame error 2 comes up

synman, 2021-12-16:
This is turning into an extensive, but very necessary, exercise.   I've brought a lot of stuff into scope here including laser detection based on $32 value, usage of Grbl 1.1 jogging for framing operations, and backwards compatibility for older devices.

I'd rather just drop support for pre 1.1 firmwares but recognize there are probably a bunch of people stuck on .9.

I'm also revisiting all embedded gcode and refactoring many of the commands used by the plugin today for various operations.

Lastly, I've finally, just moments ago, pulled trigger on a budget CNC machine.  I went with this bad boy:  https://www.foxalien.com/collections/cnc-router/products/cnc-router-machine-masuter-with-300w-spindle-bundle-kit

So yeah, anticipate some major improvements over the next month or so as I adapt the plugin to cover all my own CNC needs.

Shell

commit d5640f84 (2021-12-14): fix #44 
reworked framing gcode.  possibly fixes #35
closed 2021-12-19 by synman