synman/Octoprint-Bettergrblsupport #67: GRBL signature unknown

knucklebust0, 2022-01-17:
I had Better GRBL support previously installed and functioning, I decided to do a fresh install on octoprint on my pi 3B+, now after the fresh install in the better GRBL app in the settings under GRBL settings it shows Signature: unknown.
I have tried uninstalling it, i even tried installing an older version and still no resolve. i have attached a screen shot showing the no signature. 
Thanks in advance
[image]

synman, 2022-01-17:
It appears your machine is not connected, or has never connected.  The Grbl Settings ($$) results should be listed below the Signature (Id / Value / Description) too.  While connected you have the ability to modify them as well.

Somewhere around v2.0.0 I also change the way I identify the Grbl Signature. Previously I just scraped the Grbl Info text that shows up with a soft reset but I later changed this to use the results of $I which I query on obtaining a successful connection to the machine.

Prior method:
```
Send: \x18
Recv: 
Recv: Grbl 1.1h ['$' for help]
Recv: [MSG:'$H'|'$X' to unlock]
```

Current method (on connection):
```
Send: $I
Recv: [VER:1.1h.20201218:]
Recv: [OPT:V,35,254]
```
If after connecting to your machine these fields (including the actual Grbl Settings that should be shown below) do not populate please execute the following procedure:

1. Disconnect your machine via the Octoprint _Connection_ sidebar `Disconnect` button if it is connected.
2. Clear the Terminal buffer (`Clear All`) in the Terminal tab
3. Connect your machine via the Octoprint _Conection_ sidebar `Connect` button
4. Copy all content within the Terminal tab and provide it to me here

synman, 2022-01-26:
@knucklebust0 

this is going to get closed out if no update is provided... putting a mental timer on it.

closed 2022-02-20 by synman