synman/Octoprint-Bettergrblsupport #68: Change Jog Arrow Default value to 1.

jktplunk17, 2022-01-17:
The Default Value for a Directional Jog seems to use the max of the travel from the machine, or at least in my case it does.  On reload of octopi or switching from my iPad to computer the value will always start at 900 (my max travel).  I have on more than one occasion accidentally clicked one of the jog arrows before changing the value (fat fingers) which sends my gantry flying to either a limit switch or hard stop.  Change this to a 1 as default or allow a user specified value to be saved as the initial value. 

There may already be a way in which case I apologize, but I haven't found a way to keep it from going back to 900.
[image]

synman, 2022-01-17:
Ironically, I'm setting this to the min max of x/y based on prior feedback.   I think I can accommodate both with a setting or two within Plugin Settings.  

I have been eyeballing MPos reporting for a while now and I have a way to keep track of machine movement (easily) when it is enabled (either by clicking the "Mode" area in Grbl Control or setting $10 to 1.  I'm still working through how I want to manage this but my intent is to always know where the machine is in relation to its limits and, at the least, warn the user if bgs thinks their about to exceed them.

In the meantime, I'll get some preferences built so you can have better control over this.

closed 2022-01-18 by synman

jktplunk17, 2022-01-19:
Thanks! I greatly appreciate you working on this and getting the update out so quick!

commit 0ddf1251 (2022-01-18): fixes #71 
fixes #72
fixes #68