synman/Octoprint-Bettergrblsupport #69: failure to connect to laser

cmcfarla77, 2022-01-17:
**Describe the bug**
seems to have happened after a recent update.
Octoprint 1.7.2 
Better Grbl Support 2.1.2.dev0

repro
  Changing monitoring state from "Offline" to "Opening serial connection"
  Connecting to port /dev/ttyUSB0, baudrate 115200
  Changing monitoring state from "Opening serial connection" to "Connecting"
  Connected to: Serial<id=0xacbd8cd0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
  Send: N0 M110 N0*125
  Recv: error:7
  Recv: $0=10
  Recv: $1=25
  Recv: $2=0
  Recv: $3=0
  Recv: $4=0
  Recv: $5=0
  Recv: $6=0
  Recv: $10=1
  Recv: $11=0.010
  Recv: $12=0.002
  Recv: $13=0
  Recv: $20=0
  Recv: $21=0
  Recv: $22=0
  Recv: $23=0
  Recv: $24=25.000
  Recv: $25=500.000
  Recv: $26=250
  Recv: $27=1.000
  Recv: $30=1000
  Recv: $31=0
  Recv: $32=1
  Recv: $100=80.000
  Recv: $101=80.000
  Recv: $102=250.000
  Recv: $110=6000.000
  Recv: $111=6000.000
  Recv: $112=1000.000
  Recv: $120=1000.000
  Recv: $121=1000.000
  Recv: $122=10.000
  Recv: $130=410.000
  Recv: $131=400.000
  Recv: $132=200.000
  Recv: 
  Recv: Grbl 1.1f ['$' for help]
  No answer from the printer within the connection timeout, trying another hello
  Send: N0 M110 N0*125
  Recv: error:20

  There was a timeout while trying to connect to the printer
  Changing monitoring state from "Connecting" to "Offline"
  Connection closed, closing down monitor

synman, 2022-01-17:
Looks like we're missing an "ok" response here to tell octoprint we are good to go.   Is it possible for you to momentarily enable serial logging, capture those details while trying to connect and share them here?   I want to confirm there is nothing missing from the terminal window (there typically isn't).

        if line.startswith('Grbl'):
            # Hack to make Arduino based GRBL work.
            # When the serial port is opened, it resets and the "hello" command
            # is not processed.
            # This makes Octoprint recognise the startup message as a successful connection.
            return "ok " + line

The Grbl line should have triggered the above.

One other thing to verify is what Octoprint has configured as your "Hello" command in Printer--> Serial Connection --> Firmware & protocol --> Protocol fine tuning --> Advanced.

Will get this fixed ASAP.  Just need a little more data to figure it out.

synman, 2022-01-17:
awesome that you're already on the DEV channel.   Once I have and idea of what's going on I'll push out a commit.

synman, 2022-01-17:
triaging a couple issues here ... need to make sure they are not related.   please check out #70 and see if you have any issues in common, such as the temperature tab and gcode tab showing up.

synman, 2022-01-17:
I think I also want to see your octoprint.log file

synman, 2022-01-17:
doing some more testing on my end here....  I've got a lead on your issue... just need to figure it out.   On my Arduino laser machine, my start up looks considerably different than yours:

```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x66855ed0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: ?
Send: ?
Recv: 
Recv: Grbl 1.1g ['$' for help]
Changing monitoring state from "Connecting" to "Operational"
```

We just need to figure out why you're sending a $$ as the original announcement when I'm sending a ?

synman, 2022-01-17:
yikes... also, this does not look good... You got an Error 7 on startup:

7 EEPROM read fail. Using defaults An EEPROM read failed. Auto-restoring affected EEPROM to default values.

synman, 2022-01-18:
curious to know if 2.1.2 addresses your connection issue.  However there is something wrong with your machine you should look into getting fixed.  See #31 for more details about Grbl Error 7.

cmcfarla77, 2022-01-19:
sorry, haven't been receiving notifications from github.  I'll get the files today. Thank!

synman, 2022-01-19:
I suspect the issue related to connecting has been resolved with 2.1.3.  please be sure to update if you haven't already.

The lingering item I cannot fix is your "error: 7" condition.  This error suggests you have corrupt firmware and is not something I can solve.

cmcfarla77, 2022-01-20:
upgraded to 2.1.3 and all is good.  as to the error: 7 condition or the $$, I've been adjusting settings throughout to see if I could make it 'tick'.   I thought I had them all reverted when I did the capture above, looks like I was incorrect.  In any case, It's working now! Thanks @synman !

closed 2022-01-20 by cmcfarla77