synman/Octoprint-Bettergrblsupport #70: Ortur LM2 connexion issue

themaire, 2022-01-17:
Hi. I am Nicolas from France.

**Describe the bug**
Octoprint can't connect to my Ortur LM2. Error:20 the 1st time when the laser is fresh booter then error:9 when I retry to connect.
 
**To Reproduce**
Steps to reproduce the behavior:
 1 : Installed last OCTOPI distro -Version 0.18.0, running on Raspberry Pi 3 Model B Rev 1.2
                            Octoprint Version 1.7.2
2 : Installed Better Grbl Support (2.1.1) then rebooted the system to be "safe".
I am surprising, the temperature, control and GCODE tabs are not hidden. And, the only lines who talk about the plugin in config.yaml are ( in the plugin section ) :
  bettergrblsupport:
    _config_version: 4
    wizard_version: 2
 
3 : Go to the Octoprint's settings and discover Features > Send a checksum with the command is still in 

**Expected behavior**
--

**Screenshots**
--

**Logs**
2022-01-17 20:21:05,373 - Changing monitoring state from "Offline" to "Opening serial connection"
2022-01-17 20:21:05,379 - Connecting to port /dev/ttyACM0, baudrate 115200
2022-01-17 20:21:05,401 - Changing monitoring state from "Opening serial connection" to "Connecting"
2022-01-17 20:21:05,403 - Connected to: Serial<id=0x6c210fb0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2022-01-17 20:21:05,410 - Send: N0 M110 N0*125
2022-01-17 20:21:05,420 - Recv: error:20
2022-01-17 20:21:14,485 - Connection closed, closing down monitor
2022-01-17 20:21:14,491 - Changing monitoring state from "Connecting" to "Offline"
2022-01-17 20:21:21,854 - Changing monitoring state from "Offline" to "Opening serial connection"
2022-01-17 20:21:21,856 - Connecting to port /dev/ttyACM0, baudrate 115200
2022-01-17 20:21:21,880 - Changing monitoring state from "Opening serial connection" to "Connecting"
2022-01-17 20:21:21,882 - Connected to: Serial<id=0x6c2101f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2022-01-17 20:21:21,890 - Send: N0 M110 N0*125
2022-01-17 20:21:21,894 - Recv: 
2022-01-17 20:21:21,896 - Recv: Ortur Laser Master 2 Ready!
2022-01-17 20:21:21,898 - Recv: OLF 140.
2022-01-17 20:21:21,899 - Recv: Grbl 1.1h ['$' for help]
2022-01-17 20:21:21,901 - Recv: [MSG:'$H'|'$X' to unlock]
2022-01-17 20:21:21,902 - Recv: error:9
2022-01-17 20:21:51,932 - No answer from the printer within the connection timeout, trying another hello
2022-01-17 20:21:51,936 - Send: N0 M110 N0*125
2022-01-17 20:21:51,946 - Recv: error:9
2022-01-17 20:21:51,948 - There was a timeout while trying to connect to the printer
2022-01-17 20:21:51,953 - Changing monitoring state from "Connecting" to "Offline"
2022-01-17 20:21:51,956 - Connection closed, closing down monitor

**Additional context**
I think the plugin is not correctly installed.

Thanks a lot by your attention.
Nico

synman, 2022-01-17:
Definitely does not sound like it is installed correctly.   Is "Grbl Control" showing up as a tab?   

Do you mind sending me your octoprint.log file?  I suspect there may be an error in there I need to look at.

synman, 2022-01-17:
QQ -- by what means did you install Better Grbl Support?   Did you provide a URL or grab it from the Plugin Repository list?  If from a URL, which URL did you use?

themaire, 2022-01-17:
Hi!!!! 20 minutes to reply, it's a pleasure.

Yes the "Grbl Control" is present but not b default and is the last tab.

My octoprint.log : [https://pastebin.com/LZNP7jiZ](url)

I installed from the plugin manager > +get more button list.

Good luck. ^^

synman, 2022-01-17:
just to confirm... no URL... you just selected from the list of plugins available?

themaire, 2022-01-17:
Yes. Like as usual.

synman, 2022-01-17:
this line in your log indicates the plugin is disabled:

```
| !Better Grbl Support (2.1.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bettergrblsupport
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
```

Are you able to re-enable it in plugin manager?

themaire, 2022-01-17:
Ok. So.

It was marked as "Enabled". I clicked to disable > restarted Octoprint > enable plugin > restarted Octoprint.
Everything is at the same point.

synman, 2022-01-17:
the log file you shared was also only 60 or so lines.... do you have a complete log file you can share?

themaire, 2022-01-17:
I was the full log because the install was extra fresh. Can I try to remove the plugin and re install via the GitHub repo URL?

synman, 2022-01-17:
yes, please uninstall it... but just download it again from the list provided.  I just did a fresh install myself and was able to proceed with no issues following this procedure.

themaire, 2022-01-17:
OK.
This is the log of the plugin re-install : 
https://pastebin.com/dKxcSiVC

I have the temp, control and Code Viewer non hidden tabs.

My octoprint.log #2 : [https://pastebin.com/4T4mWUBQ](url)

octoprint config.yaml : [https://pastebin.com/VV9gKdh0](url)

Thanks

synman, 2022-01-17:
some good signs here:

```
Created wheel for Better-Grbl-Support: filename=Better_Grbl_Support-2.1.1-py3-none-any.whl size=47202 sha256=36be743e313ac295292d7e2c8b6f8b91942c25f9e254402ff0de945851846ac5
Stored in directory: /tmp/pip-ephem-wheel-cache-tojo0t3_/wheels/6b/55/f3/60967e9c94559f8db60e41cd6d9edeabea7c778940a9404ed6
Successfully built Better-Grbl-Support
Installing collected packages: Better-Grbl-Support
Successfully installed Better-Grbl-Support-2.1.1
```

Another good sign...

`|  Better Grbl Support (2.1.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bettergrblsupport`

gut tells me the last log just showed the status post install but before the required restart.... plugin is active....

some weird news....

`|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer`

this guy should be disabled...

and bingo (pardon my American English slang) .....

```
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_bettergrblsupport/__init__.py", line 203, in on_after_startup
    copyfile(src, dest)
  File "/home/pi/oprint/lib/python3.7/shutil.py", line 120, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/oprint/lib/python3.7/site-packages/octoprint_bettergrblsupport/static/txt/__bgs.profile'

```

So here's the heart of our issue.

Now I need to figure out how the default profile is missing.

synman, 2022-01-17:
do you mind showing me the output of the following command?

`ls -la /home/pi/oprint/lib/python3.7/site-packages/octoprint_bettergrblsupport/static/txt`

synman, 2022-01-17:
wait, I may not need it after all.... how in the world did I miss this:

FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/oprint/lib/python3.7/site-packages/octoprint_bettergrblsupport/static/txt/__bgs.profile'

There are two _ characters there.... the actual profile file only has one.... OMG

yup.... I screwed the pooch

will have this fixed shortly.

```
        if not os.path.exists(dest):
            src = os.path.dirname(os.path.realpath(__file__)) + os.path.sep + "static" + os.path.sep + "txt" + os.path.sep + "__bgs.profile"
            copyfile(src, dest)

```

synman, 2022-01-17:
this went directly to master.

2.1.2 should be available for install via Software Update..... click "Advanced", "force check" to accelerate the process.

closed 2022-01-17 by synman

synman, 2022-01-17:
@themaire  (Nicolas from France) I also left you an Easter egg in the release notes :)

themaire, 2022-01-18:
Hi!! Crazy, it works. You rock. Thanks you again.

Machine is able to connect and all is working fine. In the past I stoped to use you plugin because of various error:20. Now I am here to report all will goes wrong.

Have a nice day!

Nicolas.

synman, 2022-01-18:
about to release 2.1.3 with a few more bug fixes and added features

commit 768b3587 (2022-01-17): fixes #70
Possible also #69