synman/Octoprint-Bettergrblsupport #71: Standard control tab settings disappeared with the last update.

paramax55, 2022-01-17:
I have been using the plugin in a non-standard way. I set up some custom buttons for ZXY axis homing. I use octoprint to beam my .nc programs over to the pi. Then I use the custom buttons in the "control" tab to home the tool. Then I use bCNC to do the actual machining. I like the direction that this plugin appears to be going and I have been accepting all of the updates, but this last one wiped out my custom buttons. I was able to get them back with a couple hours' work, but it wasn't ideal. Can you add a warning or something if the updates are going to wipe other things?

synman, 2022-01-18:
Sorry about that.  

You should know the original plugin came bound with some custom controls that I finally got around to cleaning up.

Since they were in a global section of  config,.yaml they got wiped out as I implemented uninstall logic.

I do not recommend continued usage of the octopriht control tab.  Perhaps I can look into replicating custom controls into Grbl Control moving forward.

This would be a nice enhancement.

closed 2022-01-18 by synman

synman, 2022-01-18:
does this help ease the pain at all?

https://youtu.be/LNTXdx2BIlc

paramax55, 2022-01-18:
I think I understand what your video means, but I'm not certain. Does that mean that the buttons made in "custom control editor" show up in the grbl control tab? And, if so, will they survive updates, or do I need to copy them somewhere?

synman, 2022-01-18:
I am pulling all code related to touching anything remotely resembling a custom control in the global octoprint config with this next release (2.1.3).  I will also be including the functionality I demonstrated in the video.

It was remarkably simple to integrate it into the Grbl Control tab.   I also installed the GCode System Commands plugin and built a couple buttons for toggling my power supply.

Lastly, I'm building a wiki article noting all companion plugins I've tested / plan to ensure work moving forward.   Incidentally, custom controls is a built-in function.   Salandora's (awesome) plugins merely expose them via the UI.  I use them to this day with my 3D Printer.

paramax55, 2022-01-18:
I will look for 2.1.3 and I will look for the wiki article.

When I discovered your plugin, there was no z-axis control. But I was able to turn on the standard control tab and I discovered the custom control editor and I was happy. Octoprint let me beam my nc programs over. Your plugin let me connect to the CNC. And bCNC was the first app that ran my nc programs on the pi. And if I reboot the pi before a cut, it's flawless.

I am setting up a separate laser machine and, if your plugin runs laser programs, I will give it a try. I don't think bCNC does laser.

synman, 2022-01-18:
Here's the wiki article I started for plugin interoperability:  https://github.com/synman/Octoprint-Bettergrblsupport/wiki/Other-Plugins

I started this project about 2 (maybe 3?  time escapes me) years ago when I bought my eleksmaker laser.  I was clueless when it came to CNC.   This past December I finally bit the bullet and dived into CNC... and yeah... you can see how far it's come in the past month :)

commit 0ddf1251 (2022-01-18): fixes #71 
fixes #72
fixes #68