synman/Octoprint-Bettergrblsupport #74: Starting Position issue

d3d3d, 2022-01-20:
My Ortur Aufero LS 1 (firmware 1.87) Starting Position and Home Position is Lower Left Corner (LLC)

The plugins ver 2.1.3 and 2.2.0rc1 with starting point set to LLC behave as though the starting position is URC

Note:  Laser works correctly with LLC starting point set using current version of Lightburn. 
No issue with prior plugin until the staring point feature was added.

Thank You for developing and supporting the plugin!
Let me know how I can help.

synman, 2022-01-20:
This is related to #61  which will likely be reverted in an up and coming release.   I've decided to replace auto sensing logic with a simple "Invert" checkboxes in Plugin Settings.

I may work on this tonight.  You'll see this issue touched when I take it on.

synman, 2022-01-20:
@d3d3d I nudged this one along....  if you subscribe to the release candidate branch, please grab 2.2.0rc2 and report back on results.

See here for more details on how to switch to the RC branch:  https://github.com/synman/Octoprint-Bettergrblsupport/wiki/Bleeding-Edge-Updates

synman, 2022-01-20:
[image]

If you need to tweak axis direction while jogging / framing / probing

d3d3d, 2022-01-20:
Same issue on rc2 after inverting XYZ with LLC set & rebooting Pi.
Note that if URC is set draw frame does work correctly, but the of the direction arrows have inverted result.

Just noticed an error displayed on Octoprint and notification area when first connecting to the laser. 
The error was likely there before and I did not notice.

Grbl Error #20 Received
Unsupported command
Unsupported or invalid g-code command found in block.

Again Thank You, let me know what to do to help.

synman, 2022-01-20:
would need to see, at the least, what was in your terminal window to diagnose that #20 error.     what happens when you don't invert anything.

The way the logic is built now, 2.2.0 RC2 with no inversion set works EXACTLY like versions prior to 2.1.0.

d3d3d, 2022-01-20:
grbl error #20 occurs with and without XYZ inverted.
I can recreate error #20 by sending M876 P1 to laser via terminal.

GCODE from terminal below, I disconnected after error to capture text easier...
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyACM0@115200, /dev/ttyACM0@250000, /dev/ttyACM0@230400, /dev/ttyACM0@57600, /dev/ttyACM0@38400, /dev/ttyACM0@19200, /dev/ttyACM0@9600
Trying port /dev/ttyACM0, baudrate 115200
Connecting to port /dev/ttyACM0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0xa8a0d1f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: ?
Recv: ok
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: ?
Send: $$
Send: M876 P1
Recv: <Idle|MPos:0.000,0.000,0.000|Bf:35,7679|FS:0,0|Pn:P|CL:0,100|PS:2,1>
Recv: ok
Send: $I
Recv: <Idle|MPos:0.000,0.000,0.000|Bf:35,7680|FS:0,0|Pn:P|CL:0,100|PS:2,1>
Send: $G
Recv: ok
Recv: $0=5.0
Recv: $1=100
Recv: $2=0
Recv: $3=3
Recv: $4=7
Recv: $5=7
Recv: $6=1
Recv: $10=2047
Recv: $11=0.050
Recv: $12=0.002
Recv: $13=0
Recv: $14=6
Recv: $15=0
Recv: $16=1
Recv: $17=4
Recv: $18=0
Recv: $19=0
Recv: $20=1
Recv: $21=1
Recv: $22=1
Recv: $23=7
Recv: $24=600.000
Recv: $25=3000.000
Recv: $26=100
Recv: $27=3.000
Recv: $28=0.100
Recv: $29=0.0
Recv: $30=1000.
Recv: $31=0.
Recv: $32=1
Recv: $33=1000.000
Recv: $34=0.000
Recv: $35=0.000
Recv: $36=100.000
Recv: $37=0
Recv: $39=1
Recv: $40=1
Recv: $41=0
Recv: $42=1
Recv: $43=1
Recv: $44=3
Recv: $45=0
Recv: $46=0
Recv: $56=1.000
Recv: $57=1800.000
Recv: $58=175.000
Recv: $59=6000.000
Recv: $60=0
Recv: $61=0
Recv: $62=0
Recv: $63=3
Recv: $64=1
Recv: $65=0
Recv: $100=80.000
Recv: $101=80.000
Recv: $102=400.000
Recv: $110=6000.000
Recv: $111=6000.000
Recv: $112=1200.000
Recv: $120=1200.000
Recv: $121=800.000
Recv: $122=2200.000
Recv: $130=181.000
Recv: $131=181.000
Recv: $132=50.000
Recv: $259=0
Recv: $260=70
Recv: $261=0
Recv: $262=400
Recv: $263=30
Recv: $264=0
Recv: $265=50
Recv: $266=100
Recv: $267=0
Recv: $268=0
Recv: $269=0
Recv: $270=5120
Recv: ok
Recv: error:20
Recv: [ORIGIN: China]
Recv: [PRODUCER: ORTUR]
Recv: [AUTHOR: ORTUR]
Recv: [MODEL: Aufero 1 S2]
Recv: [OLF: 187]
Recv: [OLH: OLM_ESP_PRO_V1.2]
Recv: [SN: FBC69D2EBED425C9195DC50DE25DBD70]
Recv: [OLM:GENERAL]
Recv: [DATE:17:11:17 - Dec 15 2021]
Recv: [VER:1.1f(ESP32).20210403:]
Send: ?
Recv: [OPT:VNMZHSL,35,7677,3]
Recv: ok
Recv: [GC:G0 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F3000 S0.]
Recv: ok
Recv: <Idle|MPos:0.000,0.000,0.000|Bf:35,7680|FS:0,0|Pn:P|CL:0,100|PS:2,1|VA:24148,0|H:1,7>
Recv: ok
Send: ?
Recv: <Idle|MPos:0.000,0.000,0.000|Bf:35,7679|FS:0,0|Pn:P|CL:0,100|PS:2,1>
Recv: ok
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"

synman, 2022-01-20:
Why would you want to send an M876 command?   that is not a valid GRBL gcode command.

d3d3d, 2022-01-20:
Look at dump:  M876 P1 was sent to laser right after the $$ code when serial connection is first established...

My mistake  XYZ should NOT be inverted on Aufero LS1. 
it works as before with RC2.

synman, 2022-01-20:
You must be sending that.  It is NOT coming from BGS ...

synman, 2022-01-20:
talk about an obscure command too

https://marlinfw.org/docs/gcode/M876.html

d3d3d, 2022-01-20:
Not sure what is going on, Disabled the BGS plugin and the GCODE M876 P1 is NO longer sent at serial connection...
Will look further into this #20 error, perhaps it is Octoprint related.
Will let you know.

Please go ahead and close #74, RC2 solved the Start Position issue.

Thanks again.

synman, 2022-01-20:
do you have any other plugins installed?  maybe something left over from a prior 3d printer install?  

I'm more than happy to take a look at your octoprint.log file if you'd like it reviewed?   But yeah, I can assure you nothing in BGS would send that command.

I'm thinking another plugin is doing this... perhaps triggering off of something coming across the wire between your laser and BGS.

synman, 2022-01-20:
will close this issue but am here to help you figure out the M876 thing.   You've peaked my curiosity.

closed 2022-01-20 by synman