synman/Octoprint-Bettergrblsupport #75: New Update cannot connect to laser

mbt45, 2022-01-20:
The new upgrade is no able to connect the the laser  anymore. (Laser engraver Neje)

synman, 2022-01-20:
Please ensure you are on the latest version (2.1.3)

Additionally, if you have verified you are on 2.1.3,  I'm going to need to see your octoprint.log file.   This looks to me like there is an error being thrown and I need to see the stacktrace.

synman, 2022-01-20:
See #69 and #71

mbt45, 2022-01-20:
It is on version 2.1.3

`2022-01-20 20:26:07,019 - octoprint.startup - INFO - ******************************************************************************
2022-01-20 20:26:07,023 - octoprint.startup - INFO - Starting OctoPrint 1.7.2
2022-01-20 20:26:07,024 - octoprint.startup - INFO - ******************************************************************************
2022-01-20 20:26:11,000 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-01-20 20:26:11,001 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-01-20 20:26:11,001 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2022-01-20 20:26:11,002 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2022-01-20 20:26:11,603 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2022-01-20 20:26:11,604 - octoprint.plugin.core - WARNING - Invalid version requirement >=0.1.1,<=0.2.8 for blacklist entry GcodeEditor, ignoring
2022-01-20 20:26:11,604 - octoprint.plugin.core - WARNING - Invalid version requirement >=0.1.0,<=0.1.1 for blacklist entry gcodeleveling, ignoring
2022-01-20 20:26:11,687 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2022-01-20 20:26:13,136 - octoprint.plugin.core - INFO - Plugin GCode Viewer is disabled.
2022-01-20 20:26:13,884 - octoprint.plugin.core - INFO - Found 19 plugin(s) providing 18 mixin implementations, 42 hook handlers
2022-01-20 20:26:14,133 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2022-01-20 20:26:14,162 - octoprint.server - INFO - Intermediary server started
2022-01-20 20:26:14,193 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2022-01-20 20:26:14,247 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2022-01-20 20:26:14,252 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2022-01-20 20:26:14,253 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2022-01-20 20:26:14,254 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2022-01-20 20:26:14,254 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2022-01-20 20:26:14,255 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
2022-01-20 20:26:14,255 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2022-01-20 20:26:14,255 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2022-01-20 20:26:14,256 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2022-01-20 20:26:14,256 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2022-01-20 20:26:14,257 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: "Need(method='role', value='plugin_pi_support_check')")
2022-01-20 20:26:14,257 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2022-01-20 20:26:14,257 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_manage'), Need(method='role', value='plugin_pluginmanager_install')")
2022-01-20 20:26:14,263 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2022-01-20 20:26:14,264 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2022-01-20 20:26:14,265 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2022-01-20 20:26:15,464 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python3 -m pip" as command to invoke pip
[... cut ...]
2022-01-20 20:27:11,545 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
2022-01-20 20:27:13,549 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error"
2022-01-20 20:27:13,561 - octoprint.util.comm - INFO - Changing monitoring state from "Error" to "Offline after error"
2022-01-20 20:27:13,567 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2022-01-20 20:27:35,709 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:10.0.0.160
`

synman, 2022-01-20:
I deciphered your wall of text there.... nothing to glean from it.  I would be nice if you could just upload the file, or at the least parse it with proper carriage return / line feeds.   My eyes still hurt.

Let's set the logging level for the plugin to debug and try again.  

[image]

synman, 2022-01-20:
you can find the above in Settings --> Octoprint --> Logging

mbt45, 2022-01-21:
Sorry for the wall of text, I couldn't find a way to attach files, but I got it now
[octoprint(6).log](https://github.com/synman/Octoprint-Bettergrblsupport/files/7912789/octoprint.6.log)
.

synman, 2022-01-21:
According to the log (assuming your printer is connected to ttyUSB0 @ 115200) this error is happening at an OS level when BGS tries to inject the initial handshake.  Behind the scenes I replace the M110 with a ?.

```
2022-01-21 11:08:42,122 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyUSB0, baudrate 115200
2022-01-21 11:08:42,122 - octoprint.util.comm - INFO - Connecting to port /dev/ttyUSB0, baudrate 115200
2022-01-21 11:08:42,132 - octoprint.plugins.bettergrblsupport - DEBUG - __init__: on_event event=[PrinterStateChanged] payload=[{'state_id': 'DETECT_SERIAL', 'state_string': 'Detecting serial connection'}]
2022-01-21 11:08:42,166 - octoprint.server.util.watchdog - INFO - Running initial scan on watched folder...
2022-01-21 11:08:42,211 - octoprint.server.util.watchdog - INFO - ... initial scan done.
2022-01-21 11:08:42,243 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2022-01-21 11:08:42,270 - octoprint.plugins.bettergrblsupport - DEBUG - __init__: hook_gcode_sending phase=[sending] cmd=[M110 N0] cmd_type=[None] gcode=[M110]
2022-01-21 11:08:42,271 - octoprint.plugins.bettergrblsupport - DEBUG - Ignoring M110 N0
2022-01-21 11:08:42,309 - zeroconf - WARNING - Error with socket 5 (('10.0.10.1', 5353))): [Errno 126] Required key not available
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 987, in sendto
    self._sock.sendto(data, addr)
OSError: [Errno 126] Required key not available
```

I'll do a little bit of research into what error 126 is.  It is happening at an intrinsic level so I'm a bit curious as to what is going on.

Please confirm your machine is connected to ttyUSB0 (it likely is).   As I'm researching this, if you could also disable auto detection, set your com port and baud to ttyUSB0 / 115200, try again, and send me the log snippet that would be awesome.

synman, 2022-01-21:
I think this more accurately captures what is happening specific to auto detection.  I've ruled out the zeroconf 129 error as root cause... that's mdns like I thought, unrelated to auto negotiation of a USB connection.

```
2022-01-21 11:08:42,122 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyUSB0, baudrate 115200
2022-01-21 11:08:42,122 - octoprint.util.comm - INFO - Connecting to port /dev/ttyUSB0, baudrate 115200
2022-01-21 11:08:42,132 - octoprint.plugins.bettergrblsupport - DEBUG - __init__: on_event event=[PrinterStateChanged] payload=[{'state_id': 'DETECT_SERIAL', 'state_string': 'Detecting serial connection'}]
2022-01-21 11:08:42,243 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2022-01-21 11:08:42,270 - octoprint.plugins.bettergrblsupport - DEBUG - __init__: hook_gcode_sending phase=[sending] cmd=[M110 N0] cmd_type=[None] gcode=[M110]
2022-01-21 11:08:42,271 - octoprint.plugins.bettergrblsupport - DEBUG - Ignoring M110 N0
2022-01-21 11:08:44,268 - octoprint.plugins.bettergrblsupport - DEBUG - __init__: hook_gcode_received line=[]
2022-01-21 11:08:44,321 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #2 with timeout 2.0s
2022-01-21 11:08:44,336 - octoprint.plugins.bettergrblsupport - DEBUG - __init__: hook_gcode_sending phase=[sending] cmd=[M110 N0] cmd_type=[None] gcode=[M110]
2022-01-21 11:08:44,337 - octoprint.plugins.bettergrblsupport - DEBUG - Ignoring M110 N0
2022-01-21 11:08:46,331 - octoprint.plugins.bettergrblsupport - DEBUG - __init__: hook_gcode_received line=[]
2022-01-21 11:08:46,347 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
2022-01-21 11:08:46,363 - octoprint.plugins.bettergrblsupport - DEBUG - __init__: hook_gcode_sending phase=[sending] cmd=[M110 N0] cmd_type=[None] gcode=[M110]
2022-01-21 11:08:46,365 - octoprint.plugins.bettergrblsupport - DEBUG - Ignoring M110 N0
```
Your machine is responding with a blank response, that's what this is:

2022-01-21 11:08:44,268 - octoprint.plugins.bettergrblsupport - DEBUG - __init__: hook_gcode_received line=[]

I may have changed some code to ignore blank / empty responses.  I'm curious if this empty response is your machine's way of saying, "hey, I'm here".

What is the specific manufacturer / brand / model of your machine?

synman, 2022-01-21:
Here's the gcode sending handling logic I had prior to "cleaning" things up in 2.1.0:
```
        # suppress reset line #s
        if self.suppressM110 and cmd.upper().startswith('M110'):
            self._logger.debug('Ignoring %s', cmd)
            return ("$I", )
```
and here it is in 2.1.0+:
```
        if self.suppressM110 and cmd.upper().startswith('M110'):
            self._logger.debug('Ignoring %s', cmd)
            return ("?", )
```

I'm thinking through how I want to accommodate this for you.  My gut tells me if I return a $ command like $I (grbl version info) you'll be back in business.

synman, 2022-01-21:
@mbt45 I think I have a fix for this but would like to do some validation before elevating it the release candidate channel.  Are you up for trying the dev channel?  If so, I'll push a commit to GitHub and we can see if it solves your problem.

https://github.com/synman/Octoprint-Bettergrblsupport/wiki/Bleeding-Edge-Updates

synman, 2022-01-21:
opening this back up pending confirmation of resolution

mbt45, 2022-01-21:
I am updating. I'll let you know the result ASAP

mbt45, 2022-01-21:
Ok it solved the issue
Thank  you

synman, 2022-01-21:
also, thanks for getting me the data I needed to get to the bottom of it.  Be sure set your logging level back to INFO too.  BGS can generate some decent log data at DEBUG level.

mbt45, 2022-01-21:
Hi
I think I talked too fast.

Although it is saying connected it seem the controls don't work. And I am getting a lot of " Send: ?" in the console

Send: ?

mbt45, 2022-01-21:
[octoprint(8).log](https://github.com/synman/Octoprint-Bettergrblsupport/files/7917013/octoprint.8.log)

mbt45, 2022-01-22:
Port ttyUSB0
Autodetection  off
Model NEJE Master 2S.
Before last update I had no problem engraving with octopi better grlb

I'll be able to try Lightbrun tomorrow only.

I tried soft reset but no response
I tried $$ but not response.

It is getting pretty late here so I'll be able to see your next message tomorrow only.

Thank you again for your help

synman, 2022-01-22:
Opening this back up.  Am hopeful seeing your lightburn console output will shed some light on this.

synman, 2022-01-22:
@mbt45 I pushed out a new update to the dev channel tonight that initializes the connection with two linefeeds.  I was looking over code from other open source grbl gcode senders and this seems to be a predominant pattern.

I had to fake out Octoprint to make it work.  Octoprint intentionally filters out "blank" commands but I came up with a creative work around that consistently generates the Grbl welcome message on my two machines.

Let me know how it goes.

mbt45, 2022-01-22:
Here is the console of lightburn.

I am now reinstalling a fresh new octopi session and will download the plugin
[image]
 soon.

synman, 2022-01-22:
yeah, I fired up Lightburn last nigh as well and noted it doesn't share anything about its negotiation phase.  Before you rebuild octopi, the latest dev build of bgs.

mbt45, 2022-01-23:
Hi the last update seem to work. 

Strangely I setup a fresh new instance of octopi on a new SD card with your "stable" release of Better GRLB and it worked. I am now a little baffled. Was it is your plugin that was in cause or a conflict between an old octopi and your plugin during upgrade.

Never the less your stable release works on a fresh octopi, as well as your 4fcd97d022db261ea343293325f9b4d8dbeed603
Maybe just a warning to reinstall octopi in case they someone else the same issue as I did would be sufficient.

Thank you for your invaluable help and work.
Is there a place I can pay you a coffee ;) ?

synman, 2022-01-23:
I should have asked for a copy of your config.yaml.

You are the second person to have a weird condition that was resolved with a rebuild. 
 

One gent had a phantom M876 firing out of nowhere.  

I would've loved to have known if my /n/n fix would have made a difference here.

commit 074fba7b (2022-01-21): comm issues & others
- change default dwell to be G4 P0.01 instead of G4 P0.  this gives grbl a chance to synchronize having read some documentation
- respond to blank/empty messages with an "OK".  this should fix #75
- trigger events when saving grbl settings.  fixes an issue where the settings panel was not getting updated when grbl settings were changed elsewhere
- added support for individual X and Y axis probing
closed 2022-01-23 by synman