synman/Octoprint-Bettergrblsupport #76: Adjustable distances

on-d-mand, 2022-01-27:
**Is your feature request related to a problem? Please describe.**
When the interface loads it seems that the travel distance is set to 200 which means that I basically jog my 3018 into oblivion when I just want to move it a few mm (because I wasn't paying attention).

**Describe the solution you'd like**
A setting where the distances can be configured (comma separated nr of values) plus the default value can be set preventing unwanted large movements on any axis

**Describe alternatives you've considered**
Tried adjusting the code and managed to have option 0.1 - 50mm (enough for me) but the default stays 200 (fresh install of octopi/octoprint + plugin installation shows this behaviour for me)

**Additional idea**
Additional idea might be to have different jogging options for X/Y and Z, with just 40mm of Z-height. it's unlikely people will ever move it 50, 100 or even 200 at a time..

**PS**
I'm so glad I found your plugin, coming from 3d printing any other tools like makerverse just got me frustrated because of limitations, bugs and just not being octoprint ;-) (anyway to 'buy you a beer'?)

synman, 2022-01-27:
Take a look at #68   This was previously addressed by adding an override value in Plugin Settings.

synman, 2022-01-27:
I have a 400x380 machine... and am considering a 1000x1000 machine.   Also not all machines are oriented the same way.  My machine 0,0 is on the opposite corner of my machine than machine home (-400, -380).  For many people they like to be able to quickly move between these two points for material placement / etc.

on-d-mand, 2022-01-28:
Thanks for the quick reply! Changing the limits '[...]-axis maximum travel, millimeters' seems to fix this indeed but had me quite confused since it writes these changes to the GRBL board. Won't this affect the abilities of my machine since it should be able to move more than 50 millimeters on each command? I just don't want me to be able to do this during the positioning process to set the Work origin :-)

synman, 2022-01-28:
No need to change your limits, unless they were not previously valid :)

[image]

Initial Jog Travel is what you're looking for.

closed 2022-02-20 by synman