synman/Octoprint-Bettergrblsupport #78: Multiple Work coordinate systems

AlexonCNC, 2022-01-29:
It would be great to be able to set several work coordinate systems (i.e. G54, G55, G56, etc..) and select one of them to make it active for a specific job. A drop down menu could be added when selecting the mode, listing the various coordinate systems.

synman, 2022-01-29:
Something like this?

[image]

synman, 2022-01-29:
this is in the dev channel.... will cut a release candidate for 2.2.1 tonight and target mid week for a stable release

synman, 2022-01-29:
just realized I need update home for this to truly work...

            if direction == "home":
                self._printer.commands(["G54", "G90"])

                if axis == "X":
                    self._printer.commands(["G0 X0", "G91"])
                elif axis == "Y":
                    self._printer.commands(["G0 Y0", "G91"])
                elif axis == "Z":
                    self._printer.commands(["G0 Z0", "G91"])
                elif axis == "XY":
                    self._printer.commands(["G0 X0 Y0", "G91"])
                else:
                    self._printer.commands(["G0 X0 Y0 Z0", "G91"])

this fix (I'll take care of it a little later today) should be as simple as dropping G54" from the top of the sequence as we now have the widget for coordinate system selection

synman, 2022-01-29:
grr... forgot about G10

AlexonCNC, 2022-01-29:
Woaow ! That was quick ! Cannot help much on the code as I'm very new and on discovery to CNC. This looks promessing though ! Looking forward to see it implemented. Many thanks for this quick upgrade ! :-)

synman, 2022-01-29:
almost done.   I'm pushing these changes directly to dev channel while I'm making them so I can test on my CNC.  

I usually test locally on my laser before pushing to the dev remote but I've got my laser module precariously resting on its Y gantry right now and I'm too lazy to run downstairs to secure it.  

I'm 99% sure my next remote push will have the finished code.

AlexonCNC, 2022-01-29:
Actually, these coordinates are linked to the position of the machine. In terms of layout, I was envisionning these buttons close to the machine position values. Selecting G54 should update the X, Y, Z position numbers as it relates to the G54 0;0;0 position, right ?

AlexonCNC, 2022-01-29:
That's great - thanks !

synman, 2022-01-29:
I'm leveraging the built in GRBL work coordinate systems for this.  I previously had work origin tied to G54 and this was on my list of todos.  your request motivated me to get it out of the way.

synman, 2022-01-29:
as a best practice you job should set whatever system it wants to use in its preamble / header, but yeah, if it doesn't it will use which system is active when the job starts.

synman, 2022-01-29:
it's kind of a big deal too ... I'd been meaning to do this forever

AlexonCNC, 2022-01-29:
So, should this be parametered when you create your gcode file, from the toolpath CAM program ?

synman, 2022-01-29:
that's one way to do it if your post processor supports it.  most just use whichever system is active.

AlexonCNC, 2022-01-29:
ok.. still lots to learn !! :-)

synman, 2022-01-29:
"real" CNC operators use these systems to setup offsets for multiple copies of the same "object".  For each "copy" of the job they change the coordinate systems to offset "home".

After placing their work piece / material, they step through each system settings its "home" wherever they want that "copy" to cut.

synman, 2022-01-29:
if this pops back open you'll know I found another bug lol

closed 2022-01-29 by synman

AlexonCNC, 2022-01-29:
In any case, I was very happy to find this better grbl add-on. I use Octoprint on my 3D printer and I was desperately looking for a software that could do the same on my CNC.
Ok, I understand. In my case, I can use a milling bit or a laser on my machine. I'm planning on just swapping them depending on the job. That's why I was interested in the enhancement. And for more flexibility too...
I'll upload it when available and let you know how it works for me ! Have a good one and many thanks for your help !

synman, 2022-01-29:
you can use them however you like.  no one can judge you if you don't tell them :)

I think just about everyone who uses BGS does so because they came from the 3d printer world.  That was what motivated me to create it.

synman, 2022-01-29:
oh, one last time.... these systems tie directly to the mode status readout in the Grbl Control tab.  Clicking on Mode will toggle between WPos and MPos for coordinate reporting.  If you put your machine in WPos reporting mode (by clicking the Mode readout) you'll be able to see how each relates to its own system.  MPos will always give you your coordinates in the machine coordinate system.

AlexonCNC, 2022-01-29:
Yes, that is what I meant in an earlier comment, but I did not express myself very clearly. That's how I thought this would be working. In fact, I had initially in mind that you could toggle from MPo to G54WPo, to G55WPo, etc by clicking the Mode readout. But the way you've come up with it is perfect.

AlexonCNC, 2022-02-04:
Hi Synman,
Just wanted to thank you for this enahncement. It seems to work just perfectly !
Best regards
Alex

synman, 2022-02-04:
It had been in the "ideation" phase (translation: in my brain) for a while.  Thank you for the nudge to get it out there for all us to use!

commit ef578cde (2022-01-29): Custom Metadata & Coordinate Systems
- store calculated width/length as custom metadata when new files are uploaded
- perform dimensions calculation asynchronously
- add coordinate system widget to Grbl Control tab (and all associated plumbing) fixes #78
commit 4b27b958 (2022-01-29): fixes #78
commit 897c5880 (2022-01-29): should be the last commit to fix #78