synman/Octoprint-Bettergrblsupport #79: Auto Cooling

championwescott, 2022-02-22:
LaserGRBL implemented an "Auto Cooling" feature a few years ago.  It sends a Feed Hold and Cycle Resume command on a timer.  This helps to prevent the laser from getting too hot while in continuous operation.

Would BGS be able to implement something similar?

I can always run the laser at a lower power, but having a timed shutoff would be convenient

https://github.com/arkypita/LaserGRBL/issues/658

synman, 2022-02-22:
More than happy to take this on.

championwescott, 2022-02-22:
> More than happy to take this on. 

Awesome!

I did some quick checks by saving gcode files in LaserGRBL with the feature enabled and not enabled, but there was no change to the code.  So it appears to be implemented in the software itself.

Thanks!

synman, 2022-02-23:
It could be just as simple as triggering a "pause" in Octoprint followed by capturing the current S value and sending an S0 to shut down the laser.

When the timer is up, all that would then be required is reseting S to the last S value and resuming the job.

championwescott, 2022-02-23:
Yeah, that could certainly do it.

Obviously, making sure the laser turns off is the biggest thing, there, so it doesn't burn a hole.

Would it be better to use S0, or an M5 command?  I don't know that it would make a difference in this instance.

synman, 2022-11-04:
M5 would probably be safer and I wouldn't have to keep track of what the prior S value was

closed 2022-11-11 by synman

synman, 2022-11-11:
This will be included in 2.2.2 when it releases in a week or so

[image]

commit 6f277e2c (2022-11-11): fixes #79