synman/Octoprint-Bettergrblsupport #80: Documentation / help

n3roGit, 2022-03-05:
Hello,
I find your plugin simply ingenious. I have also found no other alternative :)

Unfortunately, there is currently little documentation for the plugin and for many things I have to guess.

Therefore, I write here times my smaller problems with my laser down. Maybe you can help here and then include this for the next in the doku.

1. after I have made PAUSE, the laser continues to run, but the power remains at 0%.
2. partly the job stops during lasering without any message in the log. The laser simply stops at X percent.
3. it would be a list of possible GCodes to use your plugin really helpful. For 3D printing you can find some. In this case, rather not.
4. sometimes Octoprint does not connect to my laser or it takes quite a long time (30 seconds) after connecting before I can send control commands to the laser.

[image]

Best regards and thank you!

synman, 2022-03-05:
There's a bit happening here.  Let me try to break it down a little.

after I have made PAUSE, the laser continues to run, but the power remains at 0%.

partly the job stops during lasering without any message in the log. The laser simply stops at X percent.

it would be a list of possible GCodes to use your plugin really helpful. For 3D printing you can find some. In this case, rather not.

I think this is a question more so about GRBL, than the plugin.  There are quite a few guides available on the internet about "GBRL GCODE".  I use the official GRBL GitHub repo as an authoritative source for my own stuff:  https://github.com/gnea/grbl/tree/master/doc/markdown

sometimes Octoprint does not connect to my laser or it takes quite a long time (30 seconds) after connecting before I can send control commands to the laser.

synman, 2022-03-05:
Do you mind sharing what your custom gcode commands "OCT01" and OCTO02" are doing?

I initially thought I would find something custom in your pause / resume gcode scripts to interfere with the pause / resume operations but it looks like you've got these two sections empty (this is a good thing).

I have been experiencing some issues of my own related to pause / resume.  Octoprint inserts some of its own gcode here (that the plugin translates to dwell) that I think may be interfering with this operation.  I think we can call this a bug that I just need to figure out and fix.

There is already code in the plugin to explicitly disable the spindle / laser on pause, and reset it on resume.  I would very much appreciate seeing your terminal window output when you experience this problem to help identify (and address) root cause.

n3roGit, 2022-03-06:
Thanks for the quick and detailed answer!
I use the OctoX command only to shut down the PI after work is done.
[image]
This seemed like the easiest way to do it.

The GRBL codes I will also take a look.

I will provide the logs as soon as possible!

I also have another question. Can it be that I do not get the laser to 100% power under Octoprint? I felt I had more power when I controlled the laser directly via Lieghtburn.

synman, 2022-03-06:
I have some similar custom commands... I do not believe yours are causing any issues either.

synman, 2022-03-11:
Well it's nice to see the code doing exactly what it is supposed to do 

M4 for dwell
Wait for idle
S0 for no power
! for feed hold

You could send a ~ here to come out of feed hold, do stuff, and go back into feed hold with another !

Resume should send:

~
S### with ### being your last power level

And then the stream would resume.

And you're not observing your laser powering off here?

n3roGit, 2022-03-11:
Hey,
unfortunately I do not understand what to do now :(

Where do I have to put something?

synman, 2022-03-11:
I don't see anything for you to put anywhere.  

Not seeing the actual pause visually, but by extrapolating what i would see, it looks like it is doing exactly what it is supposed to do.

closed 2022-11-04 by synman