synman/Octoprint-Bettergrblsupport #81: GRBL Control Tab greyed out

thommy884, 2022-03-21:
Hi there. I'm able to connect to my CNC3018pro (baudrate 115200, board with GRBL 0.8) but the grbl control tab is greyed out. If | use the normal control tab of octoprint, I can jog. I can turn on and off the spindle via terminal as well. Do u guys have an idea what I could try? Pls see the attached screenshot and octoprint logs, thanks a lot! 
[image]

[octoprint-systeminfo-20220321192109.zip](https://github.com/synman/Octoprint-Bettergrblsupport/files/8318291/octoprint-systeminfo-20220321192109.zip)

synman, 2022-03-21:
Id like to see your terminal window.  Notice how state is N/A?

thommy884, 2022-03-21:
hi. wow, you are fast! yeah, I noticed that. Not much to see in the terminal (see attached).

[image]

thommy884, 2022-03-21:
but when I jog via the ordinary control tab, then it looks like that:

[image]

synman, 2022-03-22:
The built in control tab is not supported and it's dumb luck it is working at all.  Please share your terminal window contents when attempting to connect.

I may find it in the state files you shared but yeah it would be a lot easier to just see what is being sent back and forth between Octoprint and your machine.

synman, 2022-03-22:
> hi. wow, you are fast! yeah, I noticed that. Not much to see in the terminal (see attached).
> 
> [image]
> 

Do you mind sharing what machine it is you are trying to control.  I'm guessing this is not a GRBL machine... Possibly Marlin.

thommy884, 2022-03-22:
> mind

Sure, it is this machine: https://de.aliexpress.com/item/4000096760805.html?gatewayAdapt=glo2deu&spm=a2g0o.order_list.0.0.21ef5c5f12hAcZ

This is a CNC 3018 Pro GRBL Control CNC Maschine. It works with ChiliPeppr, CNCjs or Candle. I already tested that. But Better GRBL Support would be the ideal choice for me because it is webbased, has a built in webcam support and so many plugins. 

This is shown in the terminal when I disconnect and reconnect to the CNC:

[image]

synman, 2022-03-22:
Ah... Grbl v.9

Status reporting is disabled.

Take a look at $10 ... You'll want to set it to 3.

$10=3

Report back results.

synman, 2022-03-22:
Grbl v.9 documentation is here:. https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#grbl-settings

thommy884, 2022-03-22:
Ok, so I changed $10 to 3 and this is what the terminal shows now:

[image]
[image]

thommy884, 2022-03-22:
So I restarted octoprint and the whole pi. The only thing that works within the grbl control tab is the home button (only when I previously jog via built-in control tab). I think I'll give a firmware update a try...but it is a no name board without any type of brand/indication. Should I just try to upload a firmware meant for a genmitsu 3018 pro? Or could I brick the controller in that case? Gonna do some research now. Any other suggestions? Thanks again!

synman, 2022-03-22:
do you mind sharing what your GRBL Control tab looks like now that we have $10 set to 3?   I just took a quick look at the code and if signature is empty, it should default to GRBL .9 jogging.

I'd also like you to hit the direction buttons a couple times, toggle over to the terminal window afterwards and show me what is getting logged there.

thommy884, 2022-03-23:
this is the GRBL control tab now: it is complaining all the time that "distance exceeds x axis". It's the same for y and z axis:

[image]

this is what the terminal looks like when I press the right  button on x axis:

[image]

thommy884, 2022-03-23:
Just updated to 1.1 and now i can jog, thank u so much! how can I test the spindle? M03 does not work...great plugin anyway!

synman, 2022-11-04:
M3S####

with #### being some number within your min and max spindle speed settings.

I'm closing the issue.   Fill free to open a discussion thread if you would like to discuss this further (aka not a bug :) )

closed 2022-11-04 by synman