synman/Octoprint-Bettergrblsupport #85: Missing Tabs after switching to 3d-printer

andreasthaler1966, 2022-08-01:
### Discussed in https://github.com/synman/Octoprint-Bettergrblsupport/discussions/84

<div type='discussions-op-text'>

<sup>Originally posted by **andreasthaler1966** July 31, 2022</sup>
Hi!
I just installed the GREAT plugin on my PI, wich I normally use for a 3D-Printer.
Now I wanted to switch back to the 3D-Printer (by disabeling the plugin).
Now my #control, #temp and  #gcode - tabs disappeared. 
I hope this is the correct channel to question...</div>

synman, 2022-11-04:
I think I saw this over in the discussion section.   Let me hop back over there and pull in my answer here.

synman, 2022-11-04:
https://github.com/synman/Octoprint-Bettergrblsupport/discussions/84

This is going to get the "will not fix" tag for now as I have no means of detecting when the plugin is disabled / re-enabled.

synman, 2022-11-12:
this will go out with 2.2.2 in the next week or so.  I figured out how to trap plugin disable

closed 2022-11-12 by synman

andreasthaler1966, 2022-11-12:
Super!Thank you!--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.