synman/Octoprint-Bettergrblsupport #87: GRBL Settings -- No settings appear

garyzz, 2022-11-04:
No settings appear under GRBL Settings after connecting to my CNC.  
Octoprint version 1.8.6.  Unlike a similar reported issue "GRBL signature unknown #67", I have a Signature: [VER:1.1f.20220618:] [OPT:VMZL,35,254] but no contents are listed under id, value and description.

My terminal output after clearing the terminal buffer and performing a disconnect and reconnect are posted below

**TERMINAL OUTPUT**
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyACM0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x6b83fd30, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: 

 ?
Send: 

 ?
Recv: <Idle|MPos:2.000,-15.000,0.000|Bf:35,250|FS:0,0>
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: 

 ?
Send: $$
Recv: ok
Recv: ok
Recv: <Idle|MPos:2.000,-15.000,0.000|Bf:35,251|FS:0,0>
Recv: ok
Recv: ok
Recv: ok
Recv: <Idle|MPos:2.000,-15.000,0.000|Bf:35,250|FS:0,0>
Recv: ok
Recv: ok
Recv: ok
Recv: $0=10
Recv: $1=25
Recv: $2=0
Recv: $3=6
Recv: $4=0
Recv: $5=0
Recv: $6=0
Recv: $10=115
Recv: $11=0.010
Recv: $12=0.002
Recv: $13=0
Recv: $20=0
Recv: $21=0
Recv: $22=1
Recv: $23=7
Recv: $24=25.000
Recv: $25=500.000
Recv: $26=250
Recv: $27=2.000
Recv: $30=1000
Recv: $31=0
Recv: $32=0
Recv: $38=10
Recv: $100=800.000
Recv: $101=800.000
Recv: $102=800.000
Recv: $110=2000.000
Recv: $111=2000.000
Recv: $112=100.000
Recv: $120=20.000
Recv: $121=20.000
Recv: $122=20.000
Recv: $130=500.000
Recv: $131=500.000
Recv: $132=200.000
Send: $I
Send: $G
Recv: ok
Recv: [VER:1.1f.20220618:]
Send: ?
Recv: [OPT:VMZL,35,254]
Recv: ok
Recv: [GC:G0 G54 G17 G21 G90 G94 M5 M M9 T0 F0 S0]
Recv: ok
Recv: <Idle|MPos:2.000,-15.000,0.000|Bf:35,254|FS:0,0>
Recv: ok

garyzz, 2022-11-04:
Please NOTE Possible Fix?  

--- I removed a plugin Spool Manager and now these GRBL settings have appeared ---

synman, 2022-11-04:
I'm guessing I have a knockout.js issue of some kind with that plugin.

synman, 2022-11-04:
I'm going to flag this as a bug and leave it open.   Spool Manager doesn't sound like all that vital of a plugin as part of a CNC workflow but I understand the challenge if you are trying to use the same Octoprint instance for both 3d printing and CNC workflows.

There are a number of other limitations at play too that do not make this a simple / doable thing.