synman/Octoprint-Bettergrblsupport #98: Octoprint 1.8.6 Error when installing plugin

bvaerewyck, 2022-11-08:
When installing the plugin, this error happens.

Preparing metadata (setup.py): started
DEPRECATION: Better-Grbl-Support is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559

Thank you.

synman, 2022-11-08:
I'll have to look into this.  There were some workarounds I needed to do installer wise due to a prior bug with Octoprint plugin install by commit.

Gina fixed them in 1.8.0 but I haven't gone back to undo my work around.

Stay tuned.

synman, 2022-11-08:
I'm starting to think. this isn't a 1.8.6 issue but more so something specific to PIP.   General guidance with Octoprint is to NOT update PIP and let Octoprint drive when actual update to pip is required.   I'm assuming at some point you upgraded pip, possibly via here:

[image]

I'm going to wager at this point your installation is not typical and with the version of pip you've overridden / pushed to, may experience issues with other plugins too.   

```
2022-11-08 17:20:55,331   /home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/synman/OctoPrint-Bettergrblsupport/archive/2.2.1.zip --no-cache-dir
2022-11-08 17:20:57,363 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2022-11-08 17:20:57,364 > Collecting https://github.com/synman/OctoPrint-Bettergrblsupport/archive/2.2.1.zip
2022-11-08 17:20:57,364 > Downloading https://github.com/synman/OctoPrint-Bettergrblsupport/archive/2.2.1.zip
2022-11-08 17:21:00,311 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from Better-Grbl-Support==2.2.1) (1.8.6)
...
2022-11-08 17:21:00,334 > Building wheels for collected packages: Better-Grbl-Support
2022-11-08 17:21:00,334 > Building wheel for Better-Grbl-Support (setup.py): started
2022-11-08 17:21:01,746 > Building wheel for Better-Grbl-Support (setup.py): finished with status 'done'
2022-11-08 17:21:01,747 > Created wheel for Better-Grbl-Support: filename=Better_Grbl_Support-2.2.1-py3-none-any.whl size=55902 sha256=6ec137943cb3925b42def532ce61855cef229319c8b8f4a5616d4b03908585fe
2022-11-08 17:21:01,747 > Stored in directory: /tmp/pip-ephem-wheel-cache-vle_hbvd/wheels/f1/e6/69/32463a69556b8938a3e028c109010b65a3e25b3f600b55e613
2022-11-08 17:21:01,747 > Successfully built Better-Grbl-Support
2022-11-08 17:21:01,748 > Installing collected packages: Better-Grbl-Support
2022-11-08 17:21:01,748 > Attempting uninstall: Better-Grbl-Support
2022-11-08 17:21:01,748 > Found existing installation: Better-Grbl-Support 2.2.2rc1
2022-11-08 17:21:01,748 > Uninstalling Better-Grbl-Support-2.2.2rc1:
2022-11-08 17:21:01,749 > Successfully uninstalled Better-Grbl-Support-2.2.2rc1
2022-11-08 17:21:01,749 > Successfully installed Better-Grbl-Support-2.2.1
```

synman, 2022-11-11:
Last update I'll be providing here... doing some future proofing as this incident reminded me I could revert a work-around.  But I don't think they are related.

This message is a WARNING.... PIP is not currently enforcing it and when it does, a lot more stuff is going to break

closed 2022-11-11 by synman