synman/Octoprint-Bettergrblsupport #101: version needs to be updated in release 2.2.2

paukstelis, 2022-11-20:
The old 'version loop' bug has popped up with 2.2.2. setup.py needs to be updated.

synman, 2022-11-20:
Are you in the stable or release candidate channel?

paukstelis, 2022-11-20:
stable

synman, 2022-11-20:
I'm not seeing this....

[image]

synman, 2022-11-20:
Ah, so you are stuck in 2.2.1?

paukstelis, 2022-11-20:
Yes, it says I am stuck because setup.py in the release zip file (https://github.com/synman/Octoprint-Bettergrblsupport/archive/refs/tags/2.2.2.zip) has version as 2.2.1.

paukstelis, 2022-11-20:
[image]

synman, 2022-11-20:
Did I forget to push to master?

Omg... I bet that is what it is.  And I'm fine because I pushed from RC.

Stand by.

synman, 2022-11-20:
yup... should be all better now... thanks for the reminder :)

[image]

closed 2022-11-20 by paukstelis