synman/Octoprint-Bettergrblsupport #103: Easier jogging - joystick and keyboard?

paukstelis, 2022-12-13:
The release notes suggests using USB or BT game controller for jogging. Is there some documentation on how to achieve this? I did search around but didn't find anything immediately. Alternatively, would it be possible to incorporate something like the native OctoPrint keyboard control that allows jogging when in the camera feed window?

synman, 2022-12-13:
it is literally as simple as plugging in your USB controller and / or connecting your Bluetooth controller to your client computer and then opening your browser and visiting your octoprint server.

I'm using HTML5 to facilitate this functionality: https://github.com/synman/Octoprint-Bettergrblsupport/blob/devel/octoprint_bettergrblsupport/static/js/bettergrblsupport.js#L1313

synman, 2022-12-13:
I'll look into the native keyboard controls too.   This is a decent feature request.

paukstelis, 2022-12-13:
> it is literally as simple as plugging in your USB controller and / or connecting your Bluetooth controller to your client computer and then opening your browser and visiting your octoprint server. [...]

Ahh, sounds good. I just didn't really see it referenced anywhere. I will try it out and add a wiki entry if needed.

synman, 2022-12-16:
how's it working for ya?

paukstelis, 2022-12-16:
> how's it working for ya?

I've only used it once so far, so I can't really comment much yet. It wasn't clear if the jog distance impacted the sensitivity of the joystick. X and Y certain worked. I didn't really explore if there were controls for jogging Z (do to the sensitivity issues).

synman, 2022-12-16:
no z jogging... just x and y.... and it's proportional.  it'll scale to your max rates based on how far you drive each axis on the gamepad.

paukstelis, 2022-12-16:
Yeah, just saw that once I started looking over your link above. 
I feel like the latency might have been a bit high, but I'll have to play with it again.
Would it be possible to add button events? It would be pretty slick to do +/- 1/0.1 mm Z jogs just using button presses.

paukstelis, 2022-12-17:
Answered my own question. Was quite easy to implement!

synman, 2022-12-27:
targeting keyboard support with 2.3.1

synman, 2022-12-30:
I may further extend this over the coming days.  At this point we have parity with the key mappings Octoprint itself exposes.  I feel the need to do more though :)

PR #106 (2022-12-30): Easier jogging - joystick and keyboard?
Fixes #103
commit 03125761 (2022-12-30): Easier jogging - joystick and keyboard?
Fixes #103