synman/Octoprint-Bettergrblsupport #108: Invalid Argument Error on install

Korrel307, 2023-01-05:
I am unable to install the plugin due to an invalid argument error during installation

The problem first started when I was unable to update the plugin through the normal update process on Octoprint. (I can't remember what the error was)
Believing it may solve the problem I uninstalled the plugin. This did not resolve the issue

I have since then tried to install the plugin via the Plugin Manager using the "from Plugin Repository" and "from URL" methods. All attempts resulted in the same error

This seems to only happen with this plugin as I am able to install other plugins without any errors

[plugin_softwareupdate_console.log](https://github.com/synman/Octoprint-Bettergrblsupport/files/10350570/plugin_softwareupdate_console.log)
[plugin_pluginmanager_console (1).log](https://github.com/synman/Octoprint-Bettergrblsupport/files/10350572/plugin_pluginmanager_console.1.log)

synman, 2023-01-05:
what version of python and pip are you using?

Grab me a system bundle from here:

[image]

Something go wrong with an octoprint upgrade back in December?  other plugins failed too (although they appear fine now)  You've also got some weird warnings happening for stuff:

```
2022-12-07 22:02:07,183 > Attempting uninstall: OctoPrint
2022-12-07 22:02:07,184 > Found existing installation: OctoPrint 1.8.1
2022-12-07 22:02:11,672 > Uninstalling OctoPrint-1.8.1:
2022-12-07 22:02:12,205 ! ERROR: Could not install packages due to an OSError: [WinError 32] The process cannot access the file because it is being used by another process: 'c:\\octoprint\\venv\\scripts\\octoprint.exe'
2022-12-07 22:02:12,206 ! Check the permissions.
2022-12-07 22:02:12,207 ! 
2022-12-07 22:02:12,740   C:\OctoPrint\venv\Scripts\python.exe -m pip --disable-pip-version-check install https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/2.3.0.zip --no-cache-dir
2022-12-07 22:02:17,845 ! WARNING: Ignoring invalid distribution -ctoprint (c:\octoprint\venv\lib\site-packages)
2022-12-07 22:02:17,846 ! WARNING: Ignoring invalid distribution -ctoprint (c:\octoprint\venv\lib\site-packages)
2022-12-07 22:02:18,349 > Collecting https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/2.3.0.zip
2022-12-07 22:02:20,493 > Downloading https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/2.3.0.zip
2022-12-07 22:06:28,163 > - 7.8 MB 35.7 kB/s 0:04:06
2022-12-07 22:06:28,165 > Preparing metadata (setup.py): started
2022-12-07 22:06:30,595 > Preparing metadata (setup.py): finished with status 'error'
2022-12-07 22:06:30,598 ! error: subprocess-exited-with-error
2022-12-07 22:06:30,599 ! 
2022-12-07 22:06:30,600 ! python setup.py egg_info did not run successfully.
2022-12-07 22:06:30,601 ! exit code: 4294967295
2022-12-07 22:06:30,601 ! 
2022-12-07 22:06:30,602 ! [1 lines of output]
2022-12-07 22:06:30,602 ! Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
2022-12-07 22:06:30,602 ! [end of output]
2022-12-07 22:06:30,603 ! 
2022-12-07 22:06:30,603 ! note: This error originates from a subprocess, and is likely not a problem with pip.
2022-12-07 22:06:30,604 ! error: metadata-generation-failed
2022-12-07 22:06:30,604 ! 
2022-12-07 22:06:30,605 ! Encountered error while generating package metadata.
2022-12-07 22:06:30,605 ! 
2022-12-07 22:06:30,606 ! See above for output.
2022-12-07 22:06:30,607 ! 
2022-12-07 22:06:30,607 ! note: This is an issue with the package mentioned above, not pip.
2022-12-07 22:06:30,608 ! hint: See above for details.
2022-12-07 22:06:30,626   C:\OctoPrint\venv\Scripts\python.exe -m pip --disable-pip-version-check install https://github.com/synman/OctoPrint-Bettergrblsupport/archive/2.2.2.zip --no-cache-dir
2022-12-07 22:06:40,815 ! WARNING: Ignoring invalid distribution -ctoprint (c:\octoprint\venv\lib\site-packages)
2022-12-07 22:06:40,816 ! WARNING: Ignoring invalid distribution -ctoprint (c:\octoprint\venv\lib\site-packages)
2022-12-07 22:06:41,321 > Collecting https://github.com/synman/OctoPrint-Bettergrblsupport/archive/2.2.2.zip
2022-12-07 22:06:44,354 > Downloading https://github.com/synman/OctoPrint-Bettergrblsupport/archive/2.2.2.zip
2022-12-07 22:06:47,435 > - 67.2 kB 28.7 kB/s 0:00:02
2022-12-07 22:06:47,436 > Preparing metadata (setup.py): started
2022-12-07 22:06:48,574 > Preparing metadata (setup.py): finished with status 'error'
2022-12-07 22:06:48,577 ! error: subprocess-exited-with-error
2022-12-07 22:06:48,577 ! 
2022-12-07 22:06:48,578 ! python setup.py egg_info did not run successfully.
2022-12-07 22:06:48,579 ! exit code: 4294967295
2022-12-07 22:06:48,579 ! 
2022-12-07 22:06:48,580 ! [1 lines of output]
2022-12-07 22:06:48,580 ! Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
2022-12-07 22:06:48,581 ! [end of output]
2022-12-07 22:06:48,581 ! 
2022-12-07 22:06:48,582 ! note: This error originates from a subprocess, and is likely not a problem with pip.
2022-12-07 22:06:48,582 ! error: metadata-generation-failed
2022-12-07 22:06:48,582 ! 
2022-12-07 22:06:48,583 ! Encountered error while generating package metadata.
2022-12-07 22:06:48,584 ! 
2022-12-07 22:06:48,584 ! See above for output.
```
```
2023-01-05 08:59:42,078 > Installing collected packages: OctoPrint-ipOnConnect
2023-01-05 08:59:42,078 > Running setup.py install for OctoPrint-ipOnConnect: started
2023-01-05 08:59:42,593 ! WARNING: Ignoring invalid distribution -ctoprint (c:\octoprint\venv\lib\site-packages)
2023-01-05 08:59:42,593 ! DEPRECATION: OctoPrint-ipOnConnect is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
2023-01-05 08:59:47,140 ! WARNING: Ignoring invalid distribution -ctoprint (c:\octoprint\venv\lib\site-packages)
```
Python / pip also seem to think there are prior versions installed even though they are reporting their installs failed:
```
2023-01-03 13:24:09,352 > Found existing installation: Better-Grbl-Support 2.2.1
2023-01-03 13:24:09,353 > Uninstalling Better-Grbl-Support-2.2.1:
2023-01-03 13:24:13,949 > Successfully uninstalled Better-Grbl-Support-2.2.1
```
Also, not sure if related, but worth checking out:  https://stackoverflow.com/questions/74906148/error-could-not-install-packages-due-to-an-oserror-errno-22-invalid-argument

Another novel idea is to temporarily enable DEV channel updates.   If successful, disable them and see if you can get a clean install:  https://github.com/synman/Octoprint-Bettergrblsupport/wiki/Bleeding-Edge-Updates

synman, 2023-01-05:
`023-01-05 08:10:59,417 - octoprint.util.pip - INFO - pip installs to C:\OctoPrint\venv\Lib\site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes`

Take look inside `C:\OctoPrint\venv\Lib\site-packages` and let me know if you see anything related to bettergrblsupport there.  It would likely be named "octoprint-bettergrblsupport" if it is there.  There should be quite a few subfolders in that directory (it's the source directory for every package in your virtual environment).

synman, 2023-01-05:
This may be the source of your troubles:
```
2023-01-05 08:10:59,589 - octoprint.environment - INFO - Detected environment is Python 3.10.1 under Windows (win32). Details:
|  hardware:
|    cores: 4
|    freq: 2399.0
|    ram: 6229127168
|  os:
|    bits: 64
|    id: windows
|    platform: win32
|  python:
|    pip: 22.3.1
|    version: 3.10.1
|    virtualenv: C:\OctoPrint\venv
```
I'm checking with other devs if we support 3.10.1.    Will follow up here once I get confirmation.   You may be looking at a new install here.

synman, 2023-01-05:
ok, with the help of another dev, we did find an oddity.   Python 3.10.1 should be okay, BTW.

As an initial test, try installing bgs via package manager using this URL:  https://github.com/synman/Octoprint-Bettergrblsupport/archive/refs/heads/devel.zip

if this works, I'll clean up the master branch which should fix the production release too.

synman, 2023-01-05:
possibly same root cause as #98

synman, 2023-01-06:
this has been squashed in master and rc in addition to devel

synman, 2023-01-07:
@Korrel307 would be nice to know if this solved your problem.

Korrel307, 2023-01-08:
Sorry for the delayed response. 

I was able to successfully install BGS using the URL provided above

synman, 2023-01-08:
Awesome.  This fix has also been pushed to the master branch so you should be good to go.

closed 2023-01-09 by synman