synman/Octoprint-Bettergrblsupport #116: ability to convert inches to mm

max0r4axor, 2023-04-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Some times i get gcode that was done in imperial and when its imported into octoprint bettergrblesupport it assumes its metric so an object thats lets say 1 inch by 1 inch when windowed it is 1 mm by  1 mm this also affects the movement speeds.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
The ability to change the reference measurements if it was imperial so it can convert to metric and be the correct size
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
what I have been doing is importing into a gcode software as inches and converting to mm and resaving
**Additional context**
Add any other context or screenshots about the feature request here.

synman, 2023-04-08:
Please upload a GCode file for reference and to assist with understanding the nature of the issue you are reporting.

I have to say I'm at a loss as Grbl controls the means of measurement.  I'm guessing this is an aesthetic issue as you are correct in assuming I favor metric measurements.

max0r4axor, 2023-04-08:
[cutting board stand inches and mm.zip](https://github.com/synman/Octoprint-Bettergrblsupport/files/11182543/cutting.board.stand.inches.and.mm.zip)
I uploaded an example of the gcode in both units of measurement

synman, 2023-10-05:
adding the enhancement label to this.  I may look into imperial support in a future release.  thanks for the reference gcode.  it will help in determining what all is necessary once I get to it.