synman/Octoprint-Bettergrblsupport #119: Add support for X/Y position before Z probe

rsteckler, 2023-05-31:
My shapeoko (and I'm assuming pin 5 GPIO devices) have the z probe detector at a specific location.  It would be great if the z-probe feature would auto-locate to that X/Y position before plunging.  I'm guessing two new settings:  Zprobe X position and Zprobe Y position.

Love the plugin.  Thank you so much for supporting it!

paukstelis, 2023-07-19:
I happened to be looking at probing when I saw this issue, so I took a whack at it. Not tested! Values are mm to jog before Z probing (should compensate for frame position). https://github.com/paukstelis/Octoprint-Bettergrblsupport/tree/xy_on_zprobe

synman, 2023-10-05:
Hey @paukstelis ... not sure how far your fork has diverged, but I'm more than happy to take this as a PR.

Yeah, I know... 3 months to respond.  I've been busy doing nothing else in particular these past few months.

paukstelis, 2023-10-05:
Honestly, I barely even remember this :smile: . I know I have never tested it, which makes me somewhat hesitant to submit a PR, but if someone is willing to try it out I can.