synman/Octoprint-Bettergrblsupport #127: Unable to engrave from Lightburn

magostinelli, 2024-11-15:
I export my projct from lightburn as gcode, and i upload it to octopi.

I start the process, but ir's too fast and i can't see nothing on the wood.

clevetbs, 2024-12-05:
I had the same issue. adjust the power output and speed. 
i use 195 speed with 100% for cutting on my atomstack upgraded to 40w output laser. 
for 20w i think u want 60mm/s speed

magostinelli, 2024-12-07:
I use the same settigs in lightburn software, if I connect the laser directly to che pc with lightburn, everything run as expected, but if I export the gcode (from lightburn) and import in octoprint it doesn't work, it's very strange.

clevetbs, 2024-12-07:
Did you do the grbl firmware in  audrino that it saved the settings.

magostinelli, 2024-12-07:
I don't know if my laser use arduino, in any case I don't have any firmware do update.
In case of firmware problem, I should also have problems using it directly from lightburn, but in this way it works correctly. Do you agree?

clevetbs, 2024-12-08:
https://youtu.be/a5nOei49S8I?si=XexWRQgH2ri6Ul7V
Watch this video

clevetbs, 2024-12-08:
Also turn off m7 if u using air assist...

magostinelli, 2024-12-08:
I don't have an arduinoi powered model, mine is OLM-PRO-V1.2 Motherboard,32Bits ESP32MCU.
The manufactur says don't update the firmware if everything works (is my case with lighburn directly connected).